AllskyTeam/allsky #295: Dark frame substraction continues even if DARK_FRAME_SUBTRACTION is set to false in config.sh

kimdalmeijer, 2020-12-08:
Situation is as follows:
1. A dark frame has been recorded and the /darks/ directory contains at least 1 valid dark frame.
2. DARK_FRAME_SUBTRACTION has been set to true and night captures have been made.
3. DARK_FRAME_SUBTRACTION is now set to false in config.sh (web gui or nano, does not make a difference).
4. image-processed.* keeps getting generated in ALLSKY HOME

I suggest to alter saveImageNight.sh and change:

------------------
if [ "$DARK_FRAME_SUBTRACTION" == true ] ; then
	IMAGE_TO_USE="$FILENAME-processed.$EXTENSION"
fi
------------------

for:

------------------
if [ "$DARK_FRAME_SUBTRACTION" == "true" ]; then
	IMAGE_TO_USE="$FILENAME-processed.$EXTENSION"
fi
------------------

Also a change in darkSubtract.sh; add the conditional statement.

------------------------------------------------
#!/bin/bash

cd $ALLSKY_HOME

# Subtract dark frame if there is one defined in config.sh
if [ "$DARK_FRAME_SUBTRACTION" == "true" ]; then
   # Find the closest dark frame temperature wise
   CLOSEST_TEMP=0
   DIFF=100
   for file in darks/*
   do
       if [[ -f $file ]]; then
           DARK_TEMP=$(echo $file | awk -F[/.] '{print $2}')
           DELTA=$(expr $TEMP - $CLOSEST_TEMP)
           ABS_DELTA=${DELTA#-}

           if [ "$ABS_DELTA" -lt "$DIFF"  ]; then
                   DIFF=$DELTA
                   CLOSEST_TEMP=$DARK_TEMP
           fi
       fi
   done

   if [ -e "darks/$CLOSEST_TEMP.$EXTENSION" ] ; then
           convert "$FULL_FILENAME" "darks/$CLOSEST_TEMP.$EXTENSION" -compose minus_src -composite -type TrueColor "$FILENAME-processed.$EXTENSION"
   fi
fi
------------------------------------------------

Kind regards!

thomasjacquin, 2020-12-23:
Hello Kim,
Thanks for pointing that out. I just pushed an update to the repository.

astroimagineer, 2020-12-27:
Thanks Thomas!

closed 2020-12-27 by astroimagineer