AllskyTeam/allsky #296: Camera ZWO Black&White

robertdelmas, 2020-12-12:
Hi all,
I have no AllSky camera for the moment, but I am very interested by this project.
In section "Requirements", you have listed many cameras and mainly Color Cameras (ASIxxxMC).
Is it possible to use a B&W camera as ASI178MM?
In this case, for configuration in the file "settings.json" for:
- type (0=RAW 8 bits, 1=RGB 24 bits, 2=RAW 16 bits): You choose between 0 or 2?
- wbr et wbb (these are the intensities of the red and blue components of the image): For B&W camera (0 or empty)?
For the connection to a PC/server, is it possible to use a cable or only the Wifi?
Sorry for these questions.
Have a nice day.
Robert from Toulouse (south west of France).

Alex-developer, 2020-12-12:
Hi

I use a ASI178MM in Raw16 and all works great, there is an issue I have with the Keogram and Startrails being white but I have not had chance to investigate and fix this yet. In RGB24 everything works great.

You can see some of the results here https://alexjg.co.uk/allsky/

Cheers

Alex

robertdelmas, 2020-12-13:
Hi Alex,
Thanks a lot for your answer.
I went to see you website and for example for the section "Archived Startrails", you have White pictures, after B&W pictures and blue pictures. What did you change between B&W pictures and blue pictures ?
A+
Robert

Alex-developer, 2020-12-13:
Robert

Originally I was using an ASI 120MC but stopped using it as the image quality was not very good. I swapped to the ASI 178MM which, apart from the start trails issue, is a much better camera for me.

When I get a few mins I will look at a fix for the white startrails on the 178

robertdelmas, 2020-12-13:
Thank a lot Alex for your answer.
A+
Robert

closed 2020-12-13 by robertdelmas