AllskyTeam/allsky #298: Width not divisible by two error on timelapse

ennui2342, 2020-12-15:
I get this error from ffmpeg on generating a timelapse with my asi224mc as it has an odd number of pixels. There are a number of possible fixes, see: https://stackoverflow.com/questions/60788967/ffmpeg-width-not-divisible-by-2-375x500-error In my case I added `-vf "pad=ceil(iw/2)*2:ceil(ih/2)*2"` to the arguments to ffmpeg in the script and this worked fine.

-Mark

thomasjacquin, 2020-12-23:
I used to have this error when I was resizing images while I was building the prototype for allsky with my ASI224MC. I fixed it by chopping 2 columns of pixels with image magick.
See this line https://github.com/thomasjacquin/allsky/blob/master/scripts/saveImageNight.sh#L62

The chop option does all the work.

I haven't had that issue during timelapse generation with ASI224mc though. Are you using full frame images?

ennui2342, 2021-01-03:
Good point @thomasjacquin. I guess trimming the full frame images would also do the trick, but I hadn't got to the optimisation stage of the build yet so was still working with full frame images.

Whilst there are lots of fixes, they're not immediately obvious to someone who's run into this bug when they're building the project with a sensor which has an odd number of pixels. Added the pad argument to ffmpeg would avoid the problem for those who would run into it, while not impacting everyone else except a slight performance hit.

-Mark

EricClaeys, 2021-11-06:
@ennui2342, Mark, any update?

ennui2342, 2021-11-10:
@EricClaeys Apologies, my allsky has been out of commission due to a waterlogged power supply. Once it's back up and running I'll do an upgrade and try the new parameter.

You are right in that the resolution of the 224mc is as you state and an even number. But the images generated by allsky which are used for the timelapse are 885x754 (width and height in settings are 0 for max so I'm not sure why it's smaller).

EricClaeys, 2021-11-10:
@ennui2342, sorry to hear about your power supply.  The 224MC has a native resolution of 1304x976.  You have "width" and "height" set to 0 in the WebUI, correct?
Would you please attach (don't copy/paste) your config/config.sh file?

EricClaeys, 2022-03-04:
Closing - no response from user.

closed 2022-03-04 by EricClaeys