AllskyTeam/allsky #300: No live view

Pascal78, 2020-12-24:
Dear all
I just setup today my Raspberry PI3B+ for this very nice project.
However, no image seams to be save. my http://allsky.local/ still without image.
I tried to start manually  ./allsky.sh, I can not see any error.

Thank you for your help
Pascal

***********************
Listing Attached Cameras:
- 0 ZWO ASI174MM

ZWO ASI174MM Information:
- Resolution:1936x1216
- Pixel Size: 5.9μm
- Supported Bin: 1 2
- Mono camera
- SDK version 1, 16, 0
- Initialise Camera OK
- Sensor temperature:31.000000

Capture Settings:
 Image Type: ASI_IMG_RGB24
 Resolution: 1936x1216
 Quality: 95
 Exposure: 10000ms
 Max Exposure: 20000ms
 Auto Exposure: 1
 Gain: 50
 Max Gain: 200
 Cooler Enabled: 0
 Target Temperature: 0C
 Auto Gain: 0
 Brightness: 50
 Gamma: 50
 WB Red: 53
 WB Blue: 90
 Binning: 1
 Delay: 10ms
 Daytime Delay: 5000ms
 USB Speed: 40
 Text Overlay: text
 Text Extra Filename:
 Text Extra Filename Age: 600
 Text Line Height 30px
 Text Position: 15px left, 30px top
 Font Name:  0
 Font Color: 255 , 255, 255
 Small Font Color: 0 , 0, 255
 Font Line Type: 16
 Font Size: 7.0
 Font Line: 1
 Outline Font : 0
 Flip Image: 0
 Filename: image.jpg
 Latitude: 48.8N
 Longitude: 0.7E
 Sun Elevation: -6
 Preview: 0
 Time: 1
 Darkframe: 0
 Show Details: 1
 Debug Level: 0

Saving auto exposed images every 10 ms

Press Ctrl+C to stop

thomasjacquin, 2020-12-24:
Can you see a file called `liveview-image.jpg` at this location `/home/pi/allsky`?

Pascal78, 2020-12-24:
Dear Thomas
Thank you for answering

liveview-image.jpg doesn't exist

thomasjacquin, 2020-12-24:
Can you post your `config.sh`?

Pascal78, 2020-12-24:
#!/bin/bash

# Choose between ZWO or RPiHQ or auto (prefers ZWO if present)
CAMERA="ZWO"

# Set to true to upload current image to your website
UPLOAD_IMG=false

# Set to true to upload timelapse video to your website at the end of each night
UPLOAD_VIDEO=false

# Set to true to send data to your server at the end of each night
POST_END_OF_NIGHT_DATA=false

# Set to true to generate a timelapse at the end of the night
TIMELAPSE=true

# Timelapse frame rate (number of frames per second)
FPS=25

# Set to true to generate a keogram at the end of the night (image summary of the night)
KEOGRAM=true

# Set to true to upload the keogram after it has been generated
UPLOAD_KEOGRAM=false

# Set to true to generate a startrails image of the night. Will skip brighter images to avoid over-exposure
STARTRAILS=true

# Images with a brightness higher than the threshold will be skipped for startrails image generation
BRIGHTNESS_THRESHOLD=0.1

# Set to true to upload the startrails after it has been generated
UPLOAD_STARTRAILS=false

# Set to true to enable automatic deletion of archived data (images + videos)
AUTO_DELETE=true

# Set this value to the number of archived nights you want to keep. Needs AUTO_DELETE=true to work
NIGHTS_TO_KEEP=14

# Set to true if you want to enable dark frame subtraction on your night time images
DARK_FRAME_SUBTRACTION=false

# Set to 0 to disable Daytime Capture
DAYTIME="0"

# Set 24Hr capture to true to save both night and day images to disk. By default, only night images are saved.
CAPTURE_24HR=true

# Resize images before cropping and saving. Adjust width and height according to your own sensor ratio
IMG_RESIZE=false
IMG_HEIGHT=1520
IMG_WIDTH=2028

# Crop the captured image, used to improve the images when using a fisheye lens
CROP_IMAGE=false
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
CROP_OFFSET_Y=0

# Auto stretch images saved at night
AUTO_STRETCH=false
AUTO_STRETCH_AMOUNT=10
AUTO_STRETCH_MID_POINT=10%

# Path to the camera settings (exposure, gain, delay, overlay, etc)
CAMERA_SETTINGS_DIR="/etc/raspap"

if [[ $CAMERA -eq "auto" ]]; then
  # restore currently saved autodiscovered camera mode if any
  source "$ALLSKY_HOME/autocam.sh"
fi
CAMERA_SETTINGS="$CAMERA_SETTINGS_DIR/settings_$CAMERA.json"

thomasjacquin, 2020-12-24:
Daytime capture is disabled. Is it day time at your location by any chance?

Pascal78, 2020-12-24:
I put it to '1'
But same

thomasjacquin, 2020-12-24:
Ok, try restarting the service or the RPi itself.
If it still doesn't work, run it manually and paste the output here.

There should be more lines coming after `Saving auto exposed images every 10 ms`.

Pascal78, 2020-12-24:
Nothing appears after'10ms'

*******************************

 Latitude: 48.8N
 Longitude: 0.7E
 Sun Elevation: -6
 Preview: 0
 Time: 1
 Darkframe: 0
 Show Details: 1
 Debug Level: 0

Saving auto exposed images every 10 ms

Press Ctrl+C to stop

astroimagineer, 2020-12-24:
Please run lsusb to see whether your camera doesn't back out. I suspect you have low power.  As soon as the camera tries an exposure, it resets its USB connection and nothing will happen. Sudo journalctl can tell you more.Met vriendelijke groeten, Kim DalmeijerDalmeijer ICT

Pascal78, 2020-12-25:
Thank you Thomas for this Idea,
I will try that.
I just tested my ASI 120 MM-S used for autoguiding, and I have image now, great !
Mery Christmas

Pascal

tomb18, 2021-02-09:
I have exactly the same issue but I am using a ASI120mm.  No live view
All settings are the same as above. liveview-image.jpg doesn't exist.
lsusb shows a device in the list which disappears on removal of the camera and reappears on plugging it in.
I tried restarting the service as well as rebooting.  no change.
Manually starting the allsky.sh gives this:

i@allsky:~/allsky $ ./allsky.sh
Making sure allsky.sh is not already running...
can't get debug descriptor: Resource temporarily unavailable
Trying to automatically choose between ZWO and RPI camera
ZWO camera found. Choosing ZWO
Settings check done
CAMERA: ZWO
CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
/home/pi/allsky/scripts/filename.sh: line 4: jq: command not found
Starting allsky camera...
./allsky.sh: line 67: jq: command not found
./allsky.sh: line 83: /home/pi/allsky/capture: Is a directory
pi@allsky:~/allsky $ ^C

The capture was manually added by me when it was complaining about it missing...But it didn't work before that either.
This is the very first time I use a Pi.  I did Unix 20 years ago, but am pretty rusty.
Thanks

tomb18, 2021-02-09:
Ok I figured out the jq and installed it. I also removed the folder. Manually starting allsky.sh now gives me the following:

Making sure allsky.sh is not already running...
can't get debug descriptor: Resource temporarily unavailable
Trying to automatically choose between ZWO and RPI camera
ZWO camera found. Choosing ZWO
Settings check done
CAMERA: ZWO
CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Starting allsky camera...
./allsky.sh: line 83: /home/pi/allsky/capture: No such file or directory
pi@allsky:~/allsky $ ^C

That's exactly what happened the first time but at least I no longer have the jq error.
Thanks for any help.
Tom

tomb18, 2021-02-09:
All solved.
On a new Raspberry Pi the jq command is not installed.  Installing then requires the reinstallation of the allsky program.  Once done, all is well.
Thanks looks good so far!

EricClaeys, 2021-09-16:
Closing as the issue has been resolved.

closed 2021-09-16 by EricClaeys