AllskyTeam/allsky #301: Night exposure not updating from daytime

uzelessknowledge, 2020-12-26:
I have my setup configured to capture daytime images and upload them to my website. Working fine. 
when night time comes around images are captured and stored, but the exposure is not adjusting to capture a suitable image. Auto exposure is on. 
If I go to the setting page and just click save, making no changes, then auto exposure begins to work. The next day images are fine but at night it’s the same issue. 
rPi 2
Asi120mc

uzelessknowledge, 2020-12-28:
Update: I changed the angle from -6 to 0. Using sunset seems to have resolved. At night the images are capturing with auto exposure.

rnnall, 2021-02-04:
Thanks for the update. I was having a similar issues and found in another thread. I guess my general question to the community would be is their a dedicated FAQ for stuff like this along with a community forum. I have been just reading through the Issues on **Git** to resolve most of my challenges as certain things are not clear to me for example when is set the `CAPTURE_24HR=true` why do I still need to set the angle to a different number for 24 hour capture. This is just a general observation and maybe I am not RTFM enough. Regardless the effort and time into this is greatly appreciated.

uzelessknowledge, 2021-02-04:
> Thanks for the update. I was having a similar issues and found in another thread. I guess my general question to the community would be is their a dedicated FAQ for stuff like this along with a [...]

No, no FAQ that I know of. That’s not how git works. It’s meant to be a code repo. 
there is a Facebook for allsky camera users. Some use this software and some use others. 
https://www.facebook.com/groups/172438633343696/?ref=share

EricClaeys, 2021-08-01:
A few of us are working on enhancements to allsky that among other things improves the exposures when transitioning from day to night and night to day.  No need to change the angle.
We're close to presenting it to Thomas for review.
Eric