AllskyTeam/allsky #303: Manual run of Timelapse, Startrail, & Keogram

AstroHighlander, 2020-12-27:
I was trying things out with some of the scripts and failed miserably - it's obvious I don't understand even the basics.  Also, I did not get a timelapse, startrail, or keogram compiled from last night's images.

Is there a way I can run the relevant commands (codes?) on the Pi terminal to create & save them?

Up until now I've managed to see the created files in the relevant daily directories on the Pi but I have to FTP them (Filezilla) to my website directory - I think I'm going to have to keep doing that as I just can't manage to fathom out the correct "language" to do it automatically - so disappointing for me.

AstroHighlander, 2021-01-15:
Took me ages to work out the solution, especially for transferring the time-lapse to my website.

There are very subtle differences in the coding syntax for changing or referring to directories in a RPi and a My GoDaddy hosted website (a "/" included or excluded was enough to mess it up).

Here are the two modified files in the RPi scripts directory to generate the daily (nightly) star trails, keogram, and time-lapse images and then to get them to transfer to my website images directory at the end of each night.

[endOfNight.sh.txt](https://github.com/thomasjacquin/allsky/files/5819673/endOfNight.sh.txt)
[timelapsesh..txt](https://github.com/thomasjacquin/allsky/files/5819674/timelapsesh.txt)

wramey7274, 2021-02-10:
AstroHighlander, you are correct that there are very subtle differences in coding syntax between IPSs, such as GoDaddy, and mine, BlueHost. I do wish when individuals post that they explicitly state what IPS they use for their website.

Thank you for posting these txt files.

EricClaeys, 2021-09-19:
@AstroHighlander, you shouldn't have to modify the shell scripts - editing ftp-settings.sh should be sufficient.
The most common issues people have with uploading are:
1. Certificate errors ("host not trusted", etc.).  Adding "set ssl:verify-certificate no" to /etc/lftp.conf.
2. The directories on the FTP server don't exist.  The users has to create them BEFORE trying to upload - Allsky doesn't create them.
3. The "*DIR" variables in ftp-settings.sh are incorrect.  On my server, the root is "", not "/".

Version 0.8 of Allsky has better error checking and handling so I suggest upgrading to it.  I'm working on putting the upload code from the various scripts into a new script called "upload.sh" that a user can manually execute.  It will log the output as well as what lftp commands were executed.

closed 2021-09-19 by EricClaeys