AllskyTeam/allsky #305: Timelapse issue with raw8

rankinstudio, 2020-12-27:
Using latest version, the ffmpeg timelapse script doesn't seem to work when using raw8 jpg capture. I'm guessing the image need to be debayered first. Any idea how to do this as part of the script? 

Cheers

thomasjacquin, 2020-12-29:
Hi David,

I haven't tried it on my camera but I just looked it up and I believe it should work by setting the proper pixel format `-c:v rawvideo -pix_fmt bayer_rggb8`.
See this thread for more info: https://stackoverflow.com/questions/59196725/use-ffmpeg-to-encode-raw-bayer-rggb8-frames-as-video 

Also, run `ffmpeg -pix_fmts` to list all supported input/output formats.

Let me know if that works and I'll add some logic to the timelapse script for RAW images.

rankinstudio, 2020-12-29:
Messing around with this for a bit. Doing something wrong. 

<code>
ffmpeg -y -f image2 \
        -r $FPS \
        -i images/$1/sequence/%04d.$EXTENSION \
        -b:v 2000k \
        -c:v rawvideo \
        -pix_fmt bayer_rggb8 \
        -movflags +faststart \
        images/$1/allsky-$1.mp4
</code>
Getting:
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scaler_0'

Thanks Thomas

rankinstudio, 2020-12-29:
I messed with this for hours doing every combination I could come up with in ffmpeg, no luck. I believe the issue is that "rawvideo" doesn't decoded the .jpg right.

I ended up writing my own script in python using numpy and cv2. I generate an .mp4 file with the same name as the one the allsky makes, but I get an error when trying to view it in the web panel. "No video with supported format and MIME type found". 

Cheers

thomasjacquin, 2020-12-29:
I have set my camera to RAW8 for tonight. I'll try to run ffmpeg on my side to confirm the issue.

thomasjacquin, 2020-12-30:
@rankinstudio

Hey David, here's the timelapse produced by the camera from last night. I didn't have to do anything. I'm running off the `master` branch.
http://www.thomasjacquin.com/allsky/videos/allsky-20201229.mp4

Have you tried on a small subset of frames? Maybe you have a corrupted image in the batch.

rankinstudio, 2020-12-30:
Yeah, I'm running off the master branch. No corrupted frames. Using Cv2 in python works fine. 

What camera are you using? I'm on the ASI178MC shooting 2080 x 2080 raw8 .jpg. Guessing maybe it's a difference in the bayer settings? 

This is the default ffmpeg command in the script I have. 
ffmpeg -y -f image2 \
        -r $FPS \
        -i images/$1/sequence/%04d.$EXTENSION \
        -vcodec libx264rgb \
        -b:v 2000k \
        -pix_fmt yuv420p \
        -movflags +faststart \
        images/$1/allsky-$1.mp4

When running it: 

Input #0, image2, from 'images/20201229/sequence/%04d.jpg':
  Duration: 00:01:16.16, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: mjpeg, gray(bt470bg/unknown/unknown), 2080x2080 [SAR 1:1 DAR 1:1], 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264rgb))
Press [q] to stop, [?] for help

ends up with a 0kb file.

thomasjacquin, 2020-12-30:
My codec is different.

ffmpeg -y -f image2 \
	-r $FPS \
	-i images/$1/sequence/%04d.$EXTENSION \
	-vcodec libx264 \
	-b:v 2000k \
	-pix_fmt yuv420p \
	-movflags +faststart \
	images/$1/allsky-$1.mp4

I'm using the ASI224MC. I believe the bayer pattern is RGGB.

rankinstudio, 2020-12-30:
I also tried that codec. Same results.

rankinstudio, 2020-12-30:
May be something unique to this cam. No idea. I have a working work-around though so no big deal.

thomasjacquin, 2020-12-31:
For what it's worth, here's the output of the `identify` command on an image of that night.

```
fireweed-20201229225750.jpg JPEG 1304x976 1304x976+0+0 8-bit Gray 256c 490447B 0.000u 0:00.000
```

rankinstudio, 2020-12-31:
Mine is similar. So again, no idea what the issue is.
`rankin-20201230022556.jpg JPEG 2080x2080 2080x2080+0+0 8-bit Gray 256c 1781300B 0.000u 0:00.000`

It seems like ffmpeg using the raw codec should be able to demosiac the images into color.

Here is how I do it in Python using imageio, numpy, and cv2. Takes the raw images and creates a color timelapse. (not full code, just demosiacing step)

```
        image = imageio.imread(imagesf[i], as_gray=True)
        npimage = np.uint8(image)
        rgb = cv2.cvtColor(npimage, cv2.COLOR_BAYER_GB2BGR)
```

Cheers

EricClaeys, 2021-09-19:
Closing the issue - user appears to have a viable workaround.

closed 2021-09-19 by EricClaeys