AllskyTeam/allsky #306: Time lapse file only 48kb?

meteorkees, 2020-12-31:
Running latest version on a raspberry pi 3b+ with the HQrpi cam: increased extra memory etc, but:

All looks good but the timelapse generated are only 48kb. I did a manual run. No errors occurred. Timelapse still empty.
I did also the sudo chown pi:pi -R allsky
Did not help. Any tips or suggestions?
Thanks!

dteastro, 2020-12-31:
I am having this issue with the same pi and camera., however last night I set the width and height in camera settings to 1024/768 and binning to 2X2. This morning it generated a working timelapse. I think the limited amount of ram on the pi3 is causing the issue, and making the images smaller helps.

meteorkees, 2021-01-02:
@dteastro 
thanks for your reply, but also with your settings Allsky still produces a time-lapse of 48KB. Also when I run it manually still the same, with no errors in the end. (except a empty time-lapse).

I run sudo apt-get update and - upgrade, nor results. 

Any help is welcome?

Thanks!

RobertG73, 2021-01-02:
I increased the memory swap on SD card 2048Mb and is working fine, without any issue from month, with the original PIHQ camera resolution and without binning.

CleverJay, 2021-01-09:
I'm having the same issue as @meteorkees  except my camera is an ASI120MC-S. Timelapses worked just fine till a couple weeks ago. I've since completely reinstalled the OS and Allsky. I run the timelapse.sh script with the date manually and it seems to be generating the time-lapse and echos "time-lapse completed," but the mp4 file is only 48Kb.

CleverJay, 2021-01-10:
I increased the GPU memory from 128 to 256 MB per @RobertG73 's suggestion. and Allsky rendered a time-lapse this morning. I will follow-up if this is consistent behavior.

ramonlopez71, 2021-01-12:
I have the same problem with an ASI120MC how do you change the memory of the GPU from 128 Mb to 256 Mb? I am new to Raspberry and I have no idea how to proceed. Thank you.

ramonlopez71, 2021-01-12:
In addition to not generating the timelapse, it also does not generate the startrails, only a still image of a moment of the night appears. In addition, even though I took the Darkframe shot, it does not discount the frames and they all appear with the chip noise. Any advice for all these problems? Thanks again
[image]
[image]
[image]

meteorkees, 2021-01-16:
I changed from a Pi3B+ to a Pi4 with 4GB. Still no timelaps. The saved images where all about 5MB. Even when I set in the config file the height and width to a smaller number, these settings seems no to arrive to scripts that takes the picture and save them. 
Due to the size of the images the Pi is not able to run the timelaps script properly.

Finally, when I resized the images by setting on the option Resize Images to 'true' the images are getting resized and saved to a smaller size image. With the default settings it produces now 1MB images. And the timelaps is created as it should be.

Still one question: what means and does the function width and height in the 'camera settings' menu??

robertpascale, 2021-01-24:
> Due to the size of the images the Pi is not able to run the timelaps script properly.

You can keep the stills coming through full size, but just have the FFMPEG routine resize for the timelapse. See Issue #249 for reference of what works

nacho3g, 2021-04-25:
> I increased the memory swap on SD card 2048Mb and is working fine, without any issue from months, with the original PIHQ camera resolution and without binning.

I increased the swap to 2048 MB but with the original resolution still crashing with a pi4 of 2 GB RAM. Have you made any additional adjustments?

EricClaeys, 2021-09-19:
@meteorkees, the "width" and "height" in Camera Settings determines how much of the chip is used when taking a picture.  "0" says use the whole chip.  Setting anything other than 0 effectively does a crop of the sensor from the center.  It will produce smaller images but you need to be careful - the width and height normally need to be even numbers, and with the ASI120 have to be divisible by 1024..
The "CROP" options in config.sh will also crop an image, but you can specify where the center of the crop is.  Plus, with CROP and height/width of 0, a full-size image is written, **then** it is cropped.

**All**,
version 0.8 handles creating timelapse better.  It produces messages in /var/log/allsky.log that can help determine what happened.  Please install version 0.8 (but save your old settings and images).

I have a Pi 4 with 4 GB and an ASI178 (resolution 3096x2080) that produces timelapse with no problems.  I also have a Pi 4 with 8 GB and an ASI 290 (resolution 1936x1096) and it doesn't produce timelapse unless I resize the final image.  That makes no sense to me.

What usually works for people is to try these, in this order:
1. Increase swap, especially if you have a Pi Zero or Pi 3 or earlier which have limited memory.
2. Increase GPU memory `(does anyone have a URL that describes how?)`
3. Decrease the TIMELAPSEWIDTH and TIMELAPSEHEIGHT variables in config.sh.
4. Bin 2x2.

closed 2021-09-19 by EricClaeys