AllskyTeam/allsky #307: 'allksky.local's server IP address could not be found'

ebeyonder, 2021-01-01:
I tried to access the allsky control panel but encountered the following error message:

'allksky.local's server IP address could not be found'

However, I still have internet access on the pi, and the pi is correctly delivering live view photos even as I type this.

Grateful for any help - thank you!

michastro, 2021-01-17:
Hello,
Same thing for me, I have made 'sudo gui/install.sh pi', then in my web explorer (Firefox) 'http://pi.local' doesn't work, same thing with 'http://ip-add-of-pi' .
Thanks
Michel

Alex-developer, 2021-01-21:
Whats in the /etc/avahi/avahi-daemon.conf file under the [server] section on the Pi?

EricClaeys, 2021-09-19:
Closing issue - no reply from either user in 9 months.

closed 2021-09-19 by EricClaeys