AllskyTeam/allsky #310: keogram doesn’t compile

FrankR59, 2021-01-16:
Hi Thomas,
Thanks for this awesome piece of work. 
When I try to compile keogram via „make all“ I get the following error on my RPI 3 with raspbian stretch:
keogram.cpp: In function ‘int main(int, char**)’:
keogram.cpp:141:2: error: ‘Rect2f’ is not a member of ‘cv’
  cv::Rect2f bbox = cv::RotatedRect(cv::Point2f(), imagesrc.size(), angle).boundingRect2f();
  ^~
keogram.cpp:142:25: error: ‘bbox’ was not declared in this scope
  rot.at<double>(0,2) += bbox.width/2.0 - imagesrc.cols/2.0;
                         ^~~~
Makefile:66: recipe for target 'keogram' failed
make: *** [keogram] Error 1
pi@allsky:~/allsky $ 

Can you please help?

Best regards
Frank

FrankR59, 2021-01-24:
I wonder how this version of keogram could have ever been compiled using opencv 2.4.9 which will be downloaded by the install script. "Rect2f" does not exist before oepncv 3.0.

FrankR59, 2021-01-25:
I recently managed to compile keogram by installing opencv 3.0.0 following this tutorial https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

FrankR59, 2021-01-25:
Solved by installing opencv 3

closed 2021-01-25 by FrankR59

loloderu, 2021-06-07:
I came to the same isuue and I solved the problem by replacing this line:
<<
cv::Rect2f bbox = cv::RotatedRect(cv::Point2f(), imagesrc.size(), angle).boundingRect2f();
>>
by these two:
<<
cv::RotatedRect rRect = cv::RotatedRect(cv::Point2f(), imagesrc.size(), angle);
cv::Rect bbox = rRect.boundingRect();
>>