AllskyTeam/allsky #311: Mozilla web browser issue

Jonk2, 2021-01-18:
I'm new to linux and rpis, but have years of reasonable level of Windows admin experience.

I've moved from a small pc to a rpi (model 2b for testing before I buy a modern rpi 4) for an allsky camera set up using a ZWO 178MC.

After a couple of hours of following the instructions, I managed to get an image on screen!

However, using the web server option, on 3 different Apple devices, Chrome on Win10, Edge on Win10 all appears well. Latest image shown, and updates when it should.

Mozilla Firefox however, is not ok - it appears as though the image container is permanently refreshing (calling the latest image continuously) and on a fast enough computer, it is enough to cause an epileptic fit! It's not the whole webpage, just the image.

Any ideas?

Jonk2, 2021-02-08:
Since I posted this, I have setup on an Rpi 4 and it's mostly working well.
It seems as though it is Firefox that is the issue here, and is common to most if not all images. Apparently it renders the image at the same time as displaying it, causing the 'flicker', and where the web server is set to update say every millisecond (as I think the default daytime setting is), this is horrendous to look at on Firefox.
The only way to fix it (that I've read) is to 'preload' the image in the web code prior to it being displayed. There's plenty of articles around explaining this, but I guess this has to be done in the webserver code?

EricClaeys, 2021-09-19:
@ckuethe, @thomasjacquin,
is your javascript better than mine?  This is out of my league.
Eric

ckuethe, 2021-09-20:
Can you check to see if your `daydelay` is set properly? The default is definitely not every millisecond - it's 5 seconds. https://github.com/thomasjacquin/allsky/blob/master/settings_ZWO.json.repo#L5

But in testing I see a couple of undesirable behaviors:
* in firefox, I definitely see a white flash as the frame is repainted, every `daydelay` millseconds.
* in chrome, there is no flash while the image repaints, but I'm seeing twice the daytime delay
This may be exacerbated by high latency connections (such as I might get ssh tunneling across the continent twice, over my mobile phone...)

I'm feeling a little lazy and busy; can you post a few links to the preload stuff you found @Jonk2, and I'll see if something pokes me in the eye?

Jonk2, 2021-09-20:
I can't find the original details I found before, but I have found these which backs it up. basically images need to be preloaded prior to displaying them.

https://kylekelly.com/posts/firefox-flickering/
https://stackoverflow.com/questions/36358310/load-high-res-image-in-background-firefox-flicker

The reason I set the images down to ms was I was trying to see if I could get a reasonable video stream from the camera. This isn't going to work, but even now, on FF, the image will flash at a rate of whatever the auto-refresh is set to on that particular webpage (public.php for example).

EricClaeys, 2021-09-20:
@Jonk2, are you wanting to save all the many thousands of files produced by the pseudo-video you're creating, or do you just want to view them on your Pi?  If you just want to view them, ZWO has a demo program that displays video called "main_SDK2_video".
To get it, download their SDK at [https://astronomy-imaging-camera.com/software-drivers](url) and click on the "Developers" tab, then download the Linux & Mac version to your Pi.
Run `tar -x -f ASI_linux_mac_SDK_V1.20.tar.bz2` to extract the contents, then go into the `ASI_linux_mac_SDK_V1.20/demo` directory.
Edit the Makefile.  Search for the line with "-o test_gui2_video", and at the end of it, add " -Wl,-rpath=../lib/armv7" (without the quotes).  Save the file, then  type `make platform=armv7 test_gui2_video`

Run ./test_gui2_video. and  answer the prompts.

Jonk2, 2021-09-23:
@EricClaeys no, I wanted to use the allsky software's image output as a cctv-like video feed so I could embed it in another browser window. For the daytime, there's probably a reasonable limit like 4fps or something. I essentially want to integrate all sorts of information (this allsky being one of them) on a single webpage to monitor my observatory.

At the time of testing that, was when I found the page tear effect of the image being rendered by firefox as it loads, rather than preloading it. Most other browsers as far as I know, don't show this behaviour, but it can be forced in the web code.

Anyway, thanks for the suggestion, I will have a look. I'm not sure that you can access the camera by 2 different software packages simultaneously. Maybe use that one for daytime then switch to allsky when the 'sunwait start night' is reached in the evenings.

There is another option for people interested in a video feed, and that's oacapture. Basically a linux equivalent of Sharpcap, written by a member of Stargazers Lounge (JamesF).

EricClaeys, 2022-05-18:
@Jonk2   Jon, We're not going to have time to do anything about this in Allsky, so I'm closing the issue.  Sounds like the problem is with Firefox and not other browsers.

closed 2022-05-18 by EricClaeys