AllskyTeam/allsky #312: Automate nighttime exposure length

cplonski, 2021-01-18:
I am using the RPiHQ and overall, I'm happy with it. However, during a full moon, the image looks way too bright like daytime, but during a moonless night, the images look great. Is it possible to modify the nighttime exposure length based on the illumination percentage of the moon, or some way to measure the brightness of the previous image to determine a good exposure length based on that?

hubivo, 2021-04-14:
You could use the Auto-exposure function:
"Set to 0 to disable auto-exposure at night. Auto-exposure delivers properly exposed images throughout the night even if the overall brightness of the sky changes (cloud cover, moon, aurora, etc). When set to 1, maxexposure value will be used as the delay between timelapse frames."

EricClaeys, 2021-09-19:
@hubivo, Autoeposure on the RPiHQ camera doesn't work at night as far as I can tell, or if it does, it's not very good.  The statement you quoted is only for ZWO cameras.
I think it will be very difficult to have AllSky do nighttime auto exposure for the RPiHQ since we get no feedback from the camera, unlike the ZWO cameras where we get feedback (actual exposure, gain, etc., and can adjust for the next exposure).
I believe most people simply use fixed exposures at night.  We may attempt to improve this in the future (I just put it on our possible "to do" list), but it'll be a while, hence I'm closing this issue.

closed 2021-09-19 by EricClaeys