AllskyTeam/allsky #313: Green band at the bottom of video

didiervh, 2021-01-20:
I have an issue with the mp4 video that is generated at the end of every night.  It has a broad green band that almost covers the bottom 50% of the video.  The individual images are fine, it's just the video that has this.  Any ideas?

[image]

didiervh, 2021-01-31:
Update: this happens only when viewing the mp4 file on the Raspberry Pi.  When downloading the file, all is fine.  So I believe this issue can be closed, although if someone would know the solution, that still would be great. :-)

EricClaeys, 2021-09-19:
Closing issue per user comment.  Issue isn't with AllSky software.

closed 2021-09-19 by EricClaeys