AllskyTeam/allsky #314: 'timelapse' crashes my pi

samgribley, 2021-01-24:
hi -- new user with a ZWO and allsky is fantastic. I had it running overnight last night, and early manual timelapses ran fine and looked great.  This morning I have about 3000 images, and when I run the timelapse script it runs for a while and then the pi restarts.  I have run it with the allsky service not running, and I was wondering about heat so I put some heatsinks on and then ran ffmpeg nice -20, and put on all the ffmpeg debugging.  No errors in allsky.log or syslog or from ffmpeg -- it just restarts, after about 15 seconds, and it rarely gets to 500 frames before it does.  

Any thoughts on how to debug or what the bottlenecks might be?  Thanks.

ayladehond, 2021-04-29:
Use a swap file on youre pi works great

EricClaeys, 2021-08-01:
Samgribley,
Did you try using a swap file?  If so, did it work?
Eric

pimmoon, 2021-08-10:
I use pi 3B+ and I tired with both ZWO (asi120mc) and v1.3 camera module. For me, timelapse process crashed my pi too and when I ran the timelapse script manually I noticed the pi crashed when it processed to copy up until certain image file.

So I went and change timelapse width and height in the config.sh using the GUI editor. And now it works fine. I also noticed that the image copy process completed successfully without getting stuck at certain file.

EricClaeys, 2021-08-10:
pimmoon, are you using a swap file?  If not, try using one and see if it works with full-sized files.
Eric

pimmoon, 2021-08-10:
I have not tired swap file but I really want to try. Is there any particular settings you recommend to run allsky on Pi 3B+?

Also, I plan to get a new Pi Zero WH to run allsky as well, but on Pi Zero, I plan to use basic v1.3 camera module with wide angle lense. Is there settings for Pi Zero I should be aware of? 

Thank you!

EricClaeys, 2021-08-10:
I can’t help with either question.  I’m pretty sure other people with Pi 3 are using swap files – search the issues and Google.

 

The same for the Pi Zero.  I also think others have said it’s slower, so creating a timelapse, for example, will take longer.

 

Eric

closed 2021-09-19 by EricClaeys