AllskyTeam/allsky #316: Timelapse video plays in reverse

bleara, 2021-01-28:
I am new to this fabulous project and in the process of adding an unsupported camera (just for me initially) but I was sure everything was set up right for a nights capture but when I looked at the timelapse video the stars moved right to left. I re-made the mp4 using the timelapse script and captured this debug which shows decreasing filename numbers, is this really right?  It also shows occasional frames are symlinked out of order, frame 0014 for instance. I am no gawk expert so I can't figure it out yet.

[0;32mCreating symlinks to generate timelapse
'/home/pi/allsky/images/20210125/sequence/0001.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000203.jpg'
'/home/pi/allsky/images/20210125/sequence/0002.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000202.jpg'
'/home/pi/allsky/images/20210125/sequence/0003.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000200.jpg'
'/home/pi/allsky/images/20210125/sequence/0004.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000159.jpg'
'/home/pi/allsky/images/20210125/sequence/0005.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000158.jpg'
'/home/pi/allsky/images/20210125/sequence/0006.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000156.jpg'
'/home/pi/allsky/images/20210125/sequence/0007.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000150.jpg'
'/home/pi/allsky/images/20210125/sequence/0008.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000148.jpg'
'/home/pi/allsky/images/20210125/sequence/0009.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000147.jpg'
'/home/pi/allsky/images/20210125/sequence/0010.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000146.jpg'
'/home/pi/allsky/images/20210125/sequence/0011.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000144.jpg'
'/home/pi/allsky/images/20210125/sequence/0012.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000143.jpg'
'/home/pi/allsky/images/20210125/sequence/0013.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000142.jpg'
'/home/pi/allsky/images/20210125/sequence/0014.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000139.jpg'
'/home/pi/allsky/images/20210125/sequence/0015.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000140.jpg'
'/home/pi/allsky/images/20210125/sequence/0016.jpg' -> '/home/pi/allsky/images/20210125/image-20210126000135.jpg'

ckuethe, 2021-09-17:
I think this issue can be closed since I just reprocessed 2 years of timelapses and timelapse.sh did the right thing for me.

EricClaeys, 2021-09-19:
@bleara, if this is still an issue, please install the newest version 0.8 software and see if it's still an issue.  I haven't heard of this issue before and I'm pretty confident an install of the latest code will resolve it.

closed 2021-09-19 by EricClaeys