AllskyTeam/allsky #318: Flickering/Ramping day time exposure

Jarrod191, 2021-02-05:
Hi Thomas,

Thank you for your wonderful work.

I wonder if you have managed to resolve the issue of the flickering exposures during the day time. I saw one thread where the ZWO library was replaced with an older one... but this didn't seem to help

Is this still the approach to this problem?

Regards,

Jarrod

EricClaeys, 2021-08-01:
Jarrod191,
A few of us are working on enhancements to allsky which include working around ZWO's bug with daytime exposures jumping all over the place - too dark, then fine, then too light, then too dark, etc.  ZWO is aware of the bug and it's on their "to do" list, but we'll likely come out with our workaround before they fix the problem.
Eric

Jarrod191, 2021-08-13:
Dear Eric et al

I’ve just tried the update and the work you’ve done is fantastic !  Thank you for the effort spent. 

Regards,

Dr Jarrod Koh
Director Wakefield Emergency Department

EricClaeys, 2021-08-13:
Thanks Jarrod,

It was fun making the updates with Linda and Jim and I am glad when other people find them useful.

What are your USB settings and what camera do you have?  Several people have had problems with the default of 40 and manual setting.

Eric

closed 2021-09-19 by EricClaeys