AllskyTeam/allsky #319: Desktop....

Shand359, 2021-02-07:
I feel like an id10T issue here. I have the allsky setup and working for the most part.... I have the issue where the GAIN doesn't seem to do anything regardless of settings, and that there is no desktop on the Pi showing if I need to change anything...... im a PC user so the use and navigation around the OS is very different to me. Can anyone help me thru this...? Thank you.....

Jarrod191, 2021-02-08:
Can you tell us what your setup is. 
Pi version, 
Operating system version for PiOS, 
Camera used, 
Whether you installed the "Graphical Interface" for the Allsky
How you are making changes to the Gain... via "nano settings.json" using terminal or the Graphical Interface

Jarrod

Shand359, 2021-02-08:
Oh, right.... sorry.
I'm using a rpi4 with an ASI120MC-S camera. At the moment I have the GUI installed...
I dont know anything about using the terminal, or if it even possible using Windows. Like I said, I am not well versed in using this OS.....  Frankly I'm still surprised I got it working at all....

EricClaeys, 2021-09-19:
@Shand359, FYI, in version 0.7 that you were likely using, daytime gain is hard coded at 0.  The "gain" setting was for nighttime only, and only worked if "Auto Gain" was No.
Please try with the newest version 0.8 and the newest 0.7 GUI.  If you still can't get gain to change anything, increase the "Debug Level" (go to the bottom of the Camera Settings GUI page, click on "Advanced Settings") and set "Display Gain" to Yes, then click the "Save changes" button.
The gain will now display in the image so you can make sure the program is getting the right gain.
FYI, on your camera gain goes from 0 to 100.
If this fixes your problem, please close the issue - thanks!

closed 2021-10-16 by EricClaeys