AllskyTeam/allsky #320: Change ZWO shortest exposure

Jonk2, 2021-02-08:
I cannot see if there's an easy or obvious way to change the minimum camera exposure to be less than 1ms?

My ZWO 178MC can expose at 0.032ms, and I'd like to use this for daytime settings, otherwise the image appears to be washed out.

It may be that the exposure is lower than 1ms but is displayed on the overlay as 0.001s, but the images do appear to be washed out in sunlight.

Where can I see the actual image exposure instead of the jpg overlay?

Speaking of the overlay, how can I change the 1st line to be easier to read, i.e. from 20210208 16:01:02 to something like 08-02-2021 16:01:02?

In fact, customising the overlay text would be great from within the gui side of things, rather than the default date, temperature, exposure and gain.

I'm happy to edit html and a bit of php but no more than that - but it's a case of knowing where to look and what to look for.

EricClaeys, 2021-08-01:
Jonk2,
Look in the log file (usually /var/log/allsky.log) for the microsecond exposure time.

A few of us are working on enhancements to allsky, including the ability to specify the time format (see the manual page for strftime(3).  It'll also let you specify whether or not to display the time, a one-line user-specified text line, the sensor temp, the exposure, the gain, and the brightness.  Our changes also include a workaround for a ZWO bug with daytime exposures jumping all over the place - for example, an image at 3 seconds is too dark, but an image at 2 seconds is too bright.

Eric

Jonk2, 2021-08-01:
Hi Eric,

I've looked in the log and can see the values, but this has opened a lot of questions in my head.

I didn't know there were plans for others to improve this software, and I'm really glad about that as it looked like it had gone stale.

With this in mind, I won't bring them up here unless you want me to?

I'd be happy to act as a beta tester if this helps? My PI is on a long term test indoors with nothing else besides an NTP running so I'm happy to make changes without fear of losing anything.

Jon

EricClaeys, 2021-08-03:
Jon,
I'm happy to answer any questions you have.
Eric

Jonk2, 2021-08-04:
Ok, when I look in the logfile, there's not really a lot to go on.

1. It seems that the camera is exposing all the time, but the software only 'grabs' a frame when the time is right, i.e. once every 30 seconds. Is this a function of the way the camera works? I suspect it is, but if not, there's no need to keep exposing continuously as this heats the sensor up and causes more noise and a lot of hot pixels. (I'm currently using a ZWO 178MC but looking to move to a cooled ZWO).

2. Every 30 seconds (my image refresh rate) only means 'resizing and uploading' are logged, but then I get about 150+ lines with the same timestamp showing the autoexposure and autogain numbers for the previous x number of exposures, like it's being dumped after a certain time into the log. This may be a limitation, or by design, but it doesn't make the log easy to read.

Maybe I'm overthinking it and will wait for any updates and improvements, you guys probably have a list!

EricClaeys, 2021-08-04:
Jon,
The new version starts exposing right before a picture and stops exposing right after; we've seen about a 10 degree decrease in temperature.  It also outputs a line to the logfile as it's written.  The current version buffers log output so dumps a bunch of lines at once which makes debugging hard.
If you email me at AstroEric at eccssw dot com I can send you the new code.
Eric

Jonk2, 2021-08-05:
Ok, I will drop you an email - I have looked at your forked version, but if I do a github clone from that, won't it complain that 'allsky' already exists?
Can you rename it perhaps so I can run side by side to compare, or does it not work like that?

EricClaeys, 2021-08-06:
I'm not very familiar with GitHub.  What you could do is stop the allsky process, then rename the ~/pi/allsky directory to something else, then do the gethub clone.
Eric

Jonk2, 2021-08-09:
New version has addressed this issue.

closed 2021-08-09 by Jonk2

EricClaeys, 2021-08-09:
Jon, FYI, if you want to manually set the daytime exposure to less than 1 ms, you can.  You'll need to Show Advanced Settings, then disable daytime Auto-Exposure and set the daytime Manual Exposure.  I have mine at .5 ms.  If you have your "Locale" set properly you should be able to use a comma, but try and let me know.
Eric