AllskyTeam/allsky #321: Keogram Rotation

AstroHighlander, 2021-02-09:
My camera is aligned such that the N-horizon is on right and this results in my Keogram being out by 90 degrees.  

I think there is a way to adjust for this by editing the keogram.cpp file but I don't know which parameter(s) to amend.

Can anyone help?  Thanks, Eric.

EricClaeys, 2021-09-19:
@AstroHighlander, it's easy.  In endOfNight.sh, look for the line that has `../keogram` on it.  In version 0.8 it is `../keogram $LAST_NIGHT_DIR/ $EXTENSION $OUTPUT > ${TMP}` 
Add "-rotate 270".  If that's the wrong way, use "-rotate 90".  It should look like:
`../keogram $LAST_NIGHT_DIR/ $EXTENSION $OUTPUT -rotate 270 > ${TMP}` 

FYI, the next version of AllSky will allow you to add the rotation amount to a setting in config.sh so you don't have to edit the script itself.

If this work, please close the issue.  Thanks - Eric

AstroHighlander, 2021-09-19:
Thanks for this Eric

A quick question ... how do find out which version I am using?  I've looked at the line you are referring to and it doesn't look exactly like what you've referred to - I'll copy mine below

# Generate keogram from collected images
if [[ $KEOGRAM == "true" ]]; then
        echo -e "Generating Keogram\n"
    mkdir -p $ALLSKY_HOME/images/$LAST_NIGHT/keogram/
        ../keogram $ALLSKY_HOME/images/$LAST_NIGHT/ $EXTENSION $ALLSKY_HOME/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.$EXTENSION
    if [[ $UPLOAD_KEOGRAM == "true" ]] ; then
        cd //
        cd home/pi/allsky/
        OUTPUT="/home/pi/allsky/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.$EXTENSION"
                lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST" -e "set net:max-retries 1; set net:timeout 20; set ssl:check-hostname no; cd public_html/allsky/keograms/; put $OUTPUT; bye"
    fi
        echo -e "\n"
fi

 I know nothing at all about coding so I have to be 'walked through' everything - apologies!

Eric

AstroHighlander, 2021-09-20:
Got it working!  Amended the line below

../keogram $ALLSKY_HOME/images/$LAST_NIGHT/ $EXTENSION $ALLSKY_HOME/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.$EXTENSION  -rotate 270

Thanks for your help - I'll close the issue now!

closed 2021-09-20 by AstroHighlander

EricClaeys, 2021-09-20:
@AstroHighlander, your change looks good.  You're probably running version 0.7 which is why your line looks quite a bit different than my 0.8 line.
You can determine what version you're running by looking at the README.md file or looking in /var/log/allsky.log after you start the software - one of the first lines will look like `Allsky Camera Software v0.8.`

AstroHighlander, 2021-09-20:
Thanks Eric

All is looking well & thanks for your help.

Eric