AllskyTeam/allsky #322: HQCamera shows stars in colors

WirthmU, 2021-02-10:


ayladehond, 2021-04-29:
Red balance 3.24
Blue balance 1.47

try this, is latest hq cam true balance

jcauthen78, 2021-04-29:
Changing that up this morning made a major improvement to the day-time view - looking forward to seeing what it does tonight.  

Thanks for the R,3.24, and B,1.47 suggestion!

jcauthen78, 2021-04-29:
**Before:** _R 2.2, B 2._ 
[https://jimsnerdstuff.com/allsky/examples/liveview-image_r2.2-b2.jpg](url)

**After:** _R 3.24, B 1.47_
[https://jimsnerdstuff.com/allsky/examples/liveview-image_r3.24-b1.47.jpg](url)

EricClaeys, 2021-09-19:
@WirthmU, It looks like the suggestion resolved your problem; if you agree, please close the issue.
I am working on a newer version that supports "saturation", and setting it to -100 produces a black and white image.  Hold tight.

WirthmU, 2021-09-19:
That would be great, Eric.

EricClaeys, 2021-09-20:
@WirthmU, are you running version 0.8 of Allsky?  You can see the version number after you start the software in /var/log/allsky.log.
If you are running 0.8, I can send you a TEST version of capture_RPiHQ.cpp that supports saturation for you to try.

closed 2021-10-16 by EricClaeys