AllskyTeam/allsky #324: Two continues lines on my image

pomgonewalkabout, 2021-02-14:
Sorry but I may have a silly question but what are the two continuous coloured lines across the top of my images and how do I get rid of them please?

lumdiniz, 2021-02-16:
pramim aparece eu reinicio rasp elas somem

thomasjacquin, 2021-02-16:
See this https://github.com/RobboRob/allsky/issues/10

pomgonewalkabout, 2021-03-04:
Hi Thomas,

I am enjoying playing with and learning how to use your Allsky software.

I have manually put up a few images on my website.

I live on a rural property and have the camera running from a portable battery that I can charge via a solar panel.

 

Last night some animal tipped the camera on it’s side !

 

Cheers

 

Dean 

 

https://deangale.com/stars/

https://deangale.com/blog/

 

See this RobboRob#10 <https://github.com/RobboRob/allsky/issues/10>

EricClaeys, 2021-09-17:
@pomgonewalkabout, I'm closing this issue since it's not something we have control over.  See the link @thomasjacquin provided.
FYI, the lines are actually a snapshot of exposure and/or gain graphs.  If you run raspistill manually it will run in video mode and you'll see the graphs move as time goes on.  You'll also see the exposure and gain numbers change.

closed 2021-09-17 by EricClaeys