AllskyTeam/allsky #325: Wrong variable name $TEMP in darkSubtract.sh

hoetzgit, 2021-02-17:
Hello and sorry for my bad englisch ;-)

I don't fully understand the dark image subtraction, but doesn't the $TEMP variable in the script in line 15 have to be called $DARK_TEMP?

many greetings Henry

thomasjacquin, 2021-02-17:
The TEMP variable is defined in this file: https://github.com/thomasjacquin/allsky/blob/master/scripts/darkCapture.sh#L10

It represents the current temperature. If this file doesn't exist, it uses 20C as a default.
The dark subtraction script parses all your dark files and compared the difference between current temp and the dark file. It retains the dark file that is the closest to the current temp (lowest delta).

hoetzgit, 2021-02-17:
Oh yes, now i understand. Thank you for your explanation.
Henry

closed 2021-02-17 by hoetzgit