AllskyTeam/allsky #326: All sky no longer working after trying a dark frame.

tomb18, 2021-02-21:
Hi,
The camera was working relatively fine for a nu8mber of days.  My first dark night showed a number of hot pixels so I tried to make a dark frame.
I am using the gui.  I followed the instructions here.  I covered the camera, enabled the option,  took the dark frame, then waited until the display no longer indicated a dark was being taken and then turned it off.  I then edited config.sh.  Saved everything.
Now, when it goes to take an image (right after saving or restarting...) It shows one image (the moon is out) then the next image is half the brightness.  Then the next, nothing shows up and it stays that way forever.
I reset all options to default, then rebooted, and no difference.  It will no longer make any images except for the first three.
In all case it says gain 0 and exposure -0.
Any suggestions?  Thanks, Tom

thomasjacquin, 2021-02-22:
Hi Tom,
Send your settings and I'll take a look.

tomb18, 2021-02-23:
Hi

Thanks for getting back to me. I thought I would take some time to try on my
own to get this working.

Everything had been fine until I tried to enable the dark frame subtraction
following the guide for the GUI.

After the image no longer would appear after the first 3 frames.

So I reset everything to defaults.  Rebooted a number of times.

I was about to do a complete reinstall then I decided to change the USB
bandwidth which was at 40.  I set it to 100.  Since then it all works fine.

This is very strange since I have another ZWO camera that will only work at
40.

 

Since I do have this forum, I think that there are issues with the settings
in general.  If I turn off autoexposure, and turn off auto gain, the camera
will still attempt to autoexpose or use autogain anyways.  It's been like
this since the beginning. Because of this I do not seem to be able to get a
good exposure at night.  It is always too bright.  I do see stars but the
background is much higher than I would like.

Any suggestions for a ASI120mm-s?

I notice you mention using a gain of 200 for this camera, but I can tell you
that any gain higher than 1 is not usable in the daytime.

What governs the overall brightness of the image at night?  There are some
trees in view, could this be a factor?

Thanks for your help and a nicely done  system.

Regards, Tom

dark frame. (#326)

 

GU7TANCNFSM4X7QEJ4Q> .

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

EricClaeys, 2021-04-26:
Tom,
Are you still having the problem of turning off autoexposure, and turning off auto gain, but having the camera still attempt to autoexpose or use autogain anyways?  I've seen the same thing at night with my ASI178MC (autoexposure and gain=0 will always be used during the day).

EricClaeys, 2021-08-01:
Tom ?

EricClaeys, 2021-09-17:
Closing issue - no response from user in several months.
FYI, the USB bandwidth setting impacts different cameras, even of the same model, differently.  Some people have problems with high USB values and others with low USB values.  From what I can tell it's pretty random.

closed 2021-09-17 by EricClaeys