AllskyTeam/allsky #327: Utilise a flashing record Led ?

pomgonewalkabout, 2021-02-24:
Damn.
I have my Allsky installed in a pipe running off a portable battery.
Last night it didn't record anything because maybe the power wasn't plugged in to the Pi?
Just wondering if there is a way to add a Record Led that flashes when the Allsky is recording?
Where would I add the code?  I am a evolving aging novice !

pomgonewalkabout, 2021-02-24:
I just connected a Led to a spare GPIO and using it as an external status Led. Disconnecting the internal one with an amendment in the config file.

EricClaeys, 2021-09-17:
@pomgonewalkabout, can this issue be closed?  It sounds like you implemented a LED.  Can you share whatever changes to the Allsky software you made?