AllskyTeam/allsky #328: RpiHQ auto white balance

paolobar54, 2021-02-24:
Trying to obtaining the maximum quality from my RPiHQ camera and a cheap fisheye, I noted that the setting for AutoAWB is not doing anything in the RPiHQ implementation. 
Checking the code I noted that the line for the AWB parameter parsing #633 to #639 of capture_RPiHQ.cpp are commented out and that the test for implementing the parameter at line #265 is on the asiAutoGain instead of the asiAutoAWB param.
Also what is the correlation to put the parameters for color component in the AWB code and only when the AutoGain is off?(lines 266 to 293)

I try and during the day the AWB is working well;
I'm not an expert of the use of raspistill, there is some adverse effect that is not readily apparent?

Thanks
Paolo

Empire4191, 2021-03-04:
Hello,
I have the same problem with the white balance. Unfortunately, the AWB button has no function.
In contrast, the auto gain controls the white balance. However, the images are then much too dark at night and you can no longer set the gain up by hand.
Regards,
Daniel

thomasjacquin, 2021-07-23:
Sorry, I just noticed that thread.
I have applied the patch and pushed to the master branch. It adds auto white balance to the RPiHQ camera.
Thanks @scorot for the patch.

EricClaeys, 2021-09-19:
Closing - issue resolved.

closed 2021-09-19 by EricClaeys