AllskyTeam/allsky #332: Keogram and startrails not generated - script error

acidreign666, 2021-03-08:
Hi, I am getting this error when running the endofnight.sh script manually, anyone know what is the cause of this or how to fix?

```
.....
[75/7618] image-20210307095336.jpg
[76/7618] image-20210307095341.jpg
[77/7618] image-20210307095346.jpg
[78/7618] image-20210307095351.jpg
[79/7618] image-20210307095356.jpg
[80/7618] image-20210307095401.jpg
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 140) > this->size() (which is 0)
./endOfNight.sh: line 44:  2666 Aborted                 ../keogram $ALLSKY_HOME/images/$LAST_NIGHT/ $EXTENSION $ALLSKY_HOME/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.$EXTENSION
```

Startrail gives a similar error.

hubivo, 2021-04-14:
Hi,
check the 81. picture: Is it much lighter or darker than the others maybe? Then delete it and start the script again. Do this with all other "much different" pictures.

ckuethe, 2021-10-05:
I'm thinking this is probably fixed in recent commits to add additional checks to startrails and keogram.

closed 2021-10-05 by ckuethe