AllskyTeam/allsky #335: Probleme de permission

flipper07, 2021-03-09:
Bonjour, j'ai un problème, les images et video ne s'enregistre plus.
j'ai un message d'erreur au démarrage:
Permission non accorder
Je vous joint le fichier de démarrage
Je ne comprend pas pourquoi ce matin tout fonctionner bien
Merci de votre aide

[allsky error.txt](https://github.com/thomasjacquin/allsky/files/6111676/allsky.error.txt)

flipper07, 2021-03-10:
Je repond a mon message.
Ce matin je rebranche et ça fonctionne a nouveau.
Je ne comprend pas???
si vous avez dejas eu ce problème, hier soir j'ai pourtant éteint et redémarre plusieurs fois.....

KaktuMAN, 2021-04-22:
Il faut marquer "sudo" puis la commande pour démarrer

EricClaeys, 2021-10-16:
If the errors in "allsky error.txt" are still present in version 0.8.1, please create a new Issue and attach the log file.  It looks like your permissions are bad.  The new installation script should resolve that.

closed 2021-10-16 by EricClaeys