AllskyTeam/allsky #336: Certification Problem

MichaelSkycam, 2021-03-12:
Hello,

I´ve build a Skycam using this Software and a Raspberry PI4, uploading on my Website skycam-mannheim.de .
The Skycam worked for months perfectly until my Webhoster changed the Certifiactes now. Since then the Allsky Software is no longer able to upload anything to my Webspace.
The logs are showing "Certificate verification: Not trusted".

I´ve tried the following without any success:
**ssh-keygen -R hostname
ssh user@hostname**
and
**ssh -o StrictHostKeyChecking=accept-new**
and I even deactivated StrictHostKeyChecking in ssh-config.
Nothing helps, the "Certificate verification: Not trusted"-Error remains.

Does anybody have an Idea how to fix this ?

Thanks
Michael

thomasjacquin, 2021-03-13:
Have you tried removing the previous key from your `~/.ssh/know_hosts` file?
If you are using the same domain name but the IP has changed, you'll get an error unless you remove it and re-add the new one.

MichaelSkycam, 2021-03-17:
Thanks for your Answer Thomas.
I cleared the know_hosts file and connected again to the Host I have in ftp-Settings of the Allsky-Software. After that there was a new Key in the know_hosts file, but the Error remained. The Allsky Software was not able to upload anything.

I solved the Problem by editing /etc/lftp.conf .
At the End of this file I added a Line _set ssl:verify-certificate no_ .
I don´t really understand why this worked because I have no SSL-Certificate at my Host. But however, it worked.

Thanks
Michael

EricClaeys, 2021-09-16:
Closing - issue resolved.  This fix will be added to the Wiki once I create it, as several people have had the same problem.

closed 2021-09-16 by EricClaeys