AllskyTeam/allsky #337: Automate capture in black and white at night, and color during the day

SBRCoach, 2021-03-14:
Bonjour

Je possede avec ma AllSky une ASI 224 qui est une camera couleur. La nuit je prefere faire des captures MONOCHROMES pour une meilleur qualité et moins sujette aux lumieres parasites, et couleur la journée.

* Je voudrais tout automatiser. 

Faire en sorte que la CAM passe en mode Monochrome au même moment ou les captures de nuit commencent et qu'elle repasse en mode couleur une fois que les timelapses etc... sont generés

Si quelqu'un à une petite idée ... j'ai essayé des trucs mais pas possible de copier des fichiers dans etc/raspap ...

Merci d'avance à tous et à Thomas en particulier pour son travail.

http://sbrcoach-allsky.lescigales.org

==============================

Hello

I have with my AllSky an ASI 224 which is a color camera. The night I prefer to make captures MONOCHROMES for a better quality and less prone to parasite lights, and color the day.

* I would like to automate everything. 

Make the CAM switch to Monochrome mode at the same time as the night captures start and switch back to color mode once the timelapses etc... are generated

If anyone has any idea ... I've tried some tricks but can't copy files in etc/raspap ...

Thanks in advance to all and to Thomas in particular for his work.

SBRCoach, 2021-03-15:
Je me réponds à moi même.
C'est bon j'ai trouvé la solution pour automatiser la chose.
Bonne journée

=================

I answer to myself.
It's ok, I found the solution to automate the thing.
Have a nice day

WirthmU, 2021-03-15:
It would be nice in you could share your solution!

iPadPro

SBRCoach, 2021-03-15:
Je fais un test en situation réelle la nuit prochaine, et si tout est OK je proposerais ma solution

SBRCoach, 2021-03-16:
Hi, I'm sorry, I'm posting the explanation in French. Too long to translate, and my English is too poor. Use the auto translators ... like I do in reverse.

Bonjour

Désolé je poste l'explication en Français. Trop long à traduire, utilisez les traducteurs auto ... comme moi je fais en sens inverse.

PS : Je connais à peine le stricte minimum sous Linux et depuis peu (même moins) mais j'ai réussi à faire ce que je voulais. C'est basique, ça fonctionne chez moi, comme je veux en tout cas, il y a certainement d'autres méthodes, alors soyez indulgent !!!

==> Pour les NOVICES comme moi, pensez à effectuer une sauvegarde de vos SCRIPTS avant de les modifier ...

Pour ceux que ça interesse, voici ce que j'ai fait... Si ça peu aider quelqu'un.

* Donc je rappelle que ayant une camera ZWO 224 MC en couleur et qui est définie comme camera par défaut dans l'application de Thomas "config.sh" (encore merci à lui) ... et que j'ai de la pollution lumineuse jusque 23h00 le soir, la nuit je me mettais en mode capture MONOCHROME pour atténuer cette pollution et avoir des images plus nettes

Au début je le faisait manuellement mais je voulais définir une bonne fois pour toute une façon automatique qui m'active la capture COULEUR pour la journée et passe en MONOCHROME la nuit en même temps que les captures commencent. 

Cela peu servir également pour se créer une config type pour la JOURNEE et une pour la NUIT avec des parametres specifiques par exemple.

**1/ Création d'un fichier "settings_ZWO-MONOCHROME.json" qui contiendra les parametres à activer la nuit**
Pour ce faire, A partir du AllSky Administration Panel / Camera Settings

==> Modifier le parametre "**Image Type = RAW8**"

(_Si necessaire modifiez d'autre parametres pour n'etre utilisés que la nuit_)

- Sauvegardez vos modifs
- Copiez le fichier /home/pi/allsky/settings_ZWO.json en fichier : settings_ZWO-MONOCHROME.json

**2/ Création d'un fichier "settings_ZWO-COULEUR.json" qui contiendra les parametres à activer le jour**
IDEM que pour la nuit, mais là, "Image **Type = RGB24**" 
...
- Sauvegardez vos modifs
- Copiez le fichier /home/pi/allsky/settings_ZWO.json en fichier : settings_ZWO-COULEUR.json

3/ On se retrouve désormais dans le dossier par défaut : "/home/pi/allsky/" avec 3 fichiers de configs

	- Le fichier ORIGINAL : settings_ZWO.json 
	- Le fichier NUIT / MONOCHROME : settings_ZWO-MONOCHROME.json
	- le fichier JOUR / COULEUR : settings_ZWO-COULEUR.json
	- 
4/ Maintenant on va créer un petit script basique qui va activer le fichier de NUIT / MONOCHROME : "**settings_ZWO-MONOCHROME.sh**" 

- Contenu du script crée : "/home/pi/allsky/scripts/settings_ZWO-MONOCHROME.sh"

[settings_ZWO-MONOCHROME.sh.txt](https://github.com/thomasjacquin/allsky/files/6149620/settings_ZWO-MONOCHROME.sh.txt)

5/ Ensuite le script "settings_ZWO-MONOCHROME.sh" sera lançé lors de la "bascule" en mode nuit définie dans le script déjà existant : "/home/pi/allsky/scripts/saveImageNight.sh" 

==> Code à RAJOUTER / inserer au début de "saveImageNight.sh"

#!/bin/bash
source $ALLSKY_HOME/config.sh
source $ALLSKY_HOME/scripts/filename.sh
source $ALLSKY_HOME/scripts/darkCapture.sh
source $ALLSKY_HOME/scripts/darkSubtract.sh
source $ALLSKY_HOME/scripts/ftp-settings.sh

cd //
    cd home/pi/allsky/  
	./scripts/settings_ZWO-MONOCHROME.sh

cd $ALLSKY_HOME

6/ Maintenant, création du script qui va activer le fichier de JOUR/ COULEUR: "settings_ZWO-COULEUR.sh" 

==> *** IMAGE + A telecharger : **settings_ZWO-COULEUR.sh

[settings_ZWO-COULEUR.sh.txt](https://github.com/thomasjacquin/allsky/files/6149642/settings_ZWO-COULEUR.sh.txt)
**

7/ Ensuite le script "**settings_ZWO-COULEUR.sh**" sera lançé apres la generation en fin de nuit des TIMELAPSES, STARTRAILS etc ... 

- Pour se faire, ce script va etre lancé à la fin du script déjà existant : "/home/pi/allsky/scripts/**endOfNight.sh**" ou au début du script "endOfNight_additionalSteps.sh" qui est prévu pour rajouter du code à executer APRES "endOfNight.sh"

==> Code à rajouter dans "endOfNight.sh"

cd //
    cd home/pi/allsky/  
	./scripts/settings_ZWO-COULEUR.sh

8/ IMPORTANT : Pour activer le tout, vous devez créer la variable "**DAY_NIGHT_MODE=true**" dans le fichier "**config.sh**" .... Pour désactiver toute les procedures precedentes, il suffira de Définir : DAY_NIGHT_MODE=false

9/ Voilà c'est fini - il y a certainement d'autres possibilités mais avec mes pauvres connaissances actuelles, ça me suffit, et ça marche

Si vous avez d'autres idées

Merci

closed 2021-10-16 by EricClaeys