AllskyTeam/allsky #338: the increase of exposure does not come at night after day

bortnyk2012, 2021-03-15:
Hi Thomas,

Big thanks for your project !!!

I have ZWO ASI120MM-S and Raspberry PI 4. The exposure of the frame does not increase when going  from day to night, as a result, I get dark frames, but the restart allsky service help for me.

This is the log before the restart, we can see how the exposure starts to increase  after. When switching from night mode to day mode, everything is fine.

Mar 15 19:38:41 allsky allsky.sh[659]: Auto Gain value: 50
Mar 15 19:38:41 allsky allsky.sh[659]: Auto Exposure value: 82 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Sleeping: 19928 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Exposure value: 82294 µs
Mar 15 19:38:41 allsky allsky.sh[659]: 20210315 19:38:01
Mar 15 19:38:41 allsky allsky.sh[659]: Auto Gain value: 50
Mar 15 19:38:41 allsky allsky.sh[659]: Auto Exposure value: 82 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Sleeping: 19928 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Exposure value: 82294 µs
Mar 15 19:38:41 allsky allsky.sh[659]: 20210315 19:38:21
Mar 15 19:38:41 allsky allsky.sh[659]: Auto Gain value: 50
Mar 15 19:38:41 allsky allsky.sh[659]: Auto Exposure value: 82 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Sleeping: 19928 ms
Mar 15 19:38:41 allsky allsky.sh[659]: Exposure value: 82294 µs
Mar 15 19:49:34 allsky allsky.sh[659]: 20210315 19:
Mar 15 19:49:34 allsky allsky.sh[3413]: Making sure allsky.sh is not already running...
Mar 15 19:49:34 allsky allsky.sh[3413]: can't get debug descriptor: Resource temporarily unavailable
Mar 15 19:49:34 allsky allsky.sh[3413]: Trying to automatically choose between ZWO and RPI camera
Mar 15 19:49:34 allsky allsky.sh[3413]: ZWO camera found. Choosing ZWO
Mar 15 19:49:34 allsky allsky.sh[3413]: Settings check done
Mar 15 19:49:34 allsky allsky.sh[3413]: CAMERA: ZWO
Mar 15 19:49:34 allsky allsky.sh[3413]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Mar 15 19:49:34 allsky allsky.sh[3413]: Starting allsky camera...
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[32m ******************************************
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[32m *** Allsky Camera Software v0.7 | 2020 ***
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[32m ******************************************
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[32mCapture images of the sky with a Raspberry Pi and an ASI Camera
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[33mAdd -h or -help for available options
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[0mContributors:
Mar 15 19:53:18 allsky allsky.sh[3413]: -Knut Olav Klo
Mar 15 19:53:18 allsky allsky.sh[3413]: -Daniel Johnsen
Mar 15 19:53:18 allsky allsky.sh[3413]: -Yang and Sam from ZWO
Mar 15 19:53:18 allsky allsky.sh[3413]: -Robert Wagner
Mar 15 19:53:18 allsky allsky.sh[3413]: -Michael J. Kidd - <linuxkidd@gmail.com>
Mar 15 19:53:18 allsky allsky.sh[3413]: -Chris Kuethe
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[0m
Mar 15 19:53:18 allsky allsky.sh[3413]: Listing Attached Cameras:
Mar 15 19:53:18 allsky allsky.sh[3413]: - 0 ZWO ASI120MM-S
Mar 15 19:53:18 allsky allsky.sh[3413]: ZWO ASI120MM-S Information:
Mar 15 19:53:18 allsky allsky.sh[3413]: - Resolution:1280x960
Mar 15 19:53:18 allsky allsky.sh[3413]: - Pixel Size: 3.8μm
Mar 15 19:53:18 allsky allsky.sh[3413]: - Supported Bin: 1 2
Mar 15 19:53:18 allsky allsky.sh[3413]: - Mono camera
Mar 15 19:53:18 allsky allsky.sh[3413]: - SDK version 1, 16, 3, 0
Mar 15 19:53:18 allsky allsky.sh[3413]: - Initialise Camera OK
Mar 15 19:53:18 allsky allsky.sh[3413]: - Sensor temperature:11.700000
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[32m
Mar 15 19:53:18 allsky allsky.sh[3413]: Capture Settings:
Mar 15 19:53:18 allsky allsky.sh[3413]:  Image Type: ASI_IMG_RGB24
Mar 15 19:53:18 allsky allsky.sh[3413]:  Resolution: 1280x960
Mar 15 19:53:18 allsky allsky.sh[3413]:  Quality: 95
Mar 15 19:53:18 allsky allsky.sh[3413]:  Exposure: 10000ms
Mar 15 19:53:18 allsky allsky.sh[3413]:  Max Exposure: 20000ms
Mar 15 19:53:18 allsky allsky.sh[3413]:  Auto Exposure: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Gain: 50
Mar 15 19:53:18 allsky allsky.sh[3413]:  Max Gain: 70
Mar 15 19:53:18 allsky allsky.sh[3413]:  Cooler Enabled: 0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Target Temperature: 0C
Mar 15 19:53:18 allsky allsky.sh[3413]:  Auto Gain: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Brightness: 50
Mar 15 19:53:18 allsky allsky.sh[3413]:  Gamma: 50
Mar 15 19:53:18 allsky allsky.sh[3413]:  WB Red: 53
Mar 15 19:53:18 allsky allsky.sh[3413]:  WB Blue: 90
Mar 15 19:53:18 allsky allsky.sh[3413]:  Binning: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Delay: 10ms
Mar 15 19:53:18 allsky allsky.sh[3413]:  Daytime Delay: 5000ms
Mar 15 19:53:18 allsky allsky.sh[3413]:  USB Speed: 40
Mar 15 19:53:18 allsky allsky.sh[3413]:  Text Overlay: text
Mar 15 19:53:18 allsky allsky.sh[3413]:  Text Extra Filename:
Mar 15 19:53:18 allsky allsky.sh[3413]:  Text Extra Filename Age: 600
Mar 15 19:53:18 allsky allsky.sh[3413]:  Text Line Height 30px
Mar 15 19:53:18 allsky allsky.sh[3413]:  Text Position: 15px left, 30px top
Mar 15 19:53:18 allsky allsky.sh[3413]:  Font Name:  0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Font Color: 255 , 255, 255
Mar 15 19:53:18 allsky allsky.sh[3413]:  Small Font Color: 0 , 0, 255
Mar 15 19:53:18 allsky allsky.sh[3413]:  Font Line Type: 16
Mar 15 19:53:18 allsky allsky.sh[3413]:  Font Size: 7.0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Font Line: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Outline Font : 0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Flip Image: 0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Filename: image.jpg
Mar 15 19:53:18 allsky allsky.sh[3413]:  Latitude: 44.73N
Mar 15 19:53:18 allsky allsky.sh[3413]:  Longitude: 34.01E
Mar 15 19:53:18 allsky allsky.sh[3413]:  Sun Elevation: -6
Mar 15 19:53:18 allsky allsky.sh[3413]:  Preview: 0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Time: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Darkframe: 0
Mar 15 19:53:18 allsky allsky.sh[3413]:  Show Details: 1
Mar 15 19:53:18 allsky allsky.sh[3413]:  Debug Level: 0
Mar 15 19:53:18 allsky allsky.sh[3413]: #033[0m
Mar 15 19:53:18 allsky allsky.sh[3413]: Saving auto exposed images every 10 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Press Ctrl+C to stop
Mar 15 19:53:18 allsky allsky.sh[3413]: Exposure value: 10000000 µs
Mar 15 19:53:18 allsky allsky.sh[3413]: 20210315 19:49:47
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Gain value: 53
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Exposure value: 10000 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Sleeping: 10010 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Exposure value: 10000000 µs
Mar 15 19:53:18 allsky allsky.sh[3413]: 20210315 19:49:57
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Gain value: 56
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Exposure value: 10000 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Sleeping: 10010 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Exposure value: 10000000 µs
Mar 15 19:53:18 allsky allsky.sh[3413]: 20210315 19:50:07
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Gain value: 59
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Exposure value: 10000 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Sleeping: 10010 ms
Mar 15 19:53:18 allsky allsky.sh[3413]: Exposure value: 10000000 µs
Mar 15 19:53:18 allsky allsky.sh[3413]: 20210315 19:50:17
Mar 15 19:53:18 allsky allsky.sh[3413]: Auto Gain value: 62

[image]
[image]

Thx for your help!
Alex.

thomasjacquin, 2021-03-16:
Looks like you have auto-gain and auto-exposure turned on. They tend to fight when they're both on so I would disable one of them (I only use auto-exposure).

hubivo, 2021-04-14:
Hi @ all,

I have the same problem with not longer automatically starting of auto-exposure with beginning of night time capture. I don't use auto-gain. This problem is also described in https://github.com/thomasjacquin/allsky/issues/301. The problem was starting after a long time running the allsky without any issues (more than 8 months). If I do as described in #301: Go to the camera setting page and just click "Save changes" without making changes, then auto-exposure starts working. But I don't want to check every evening and do that.
I tried already to update the Raspi and used the newest release of Allsky-software, but there are now other problems (e.g. imaging at daytime stops working after ca. 2 minutes...), so I restored my old image that I have the same status from June 2020.
Any suggestions to the auto-exposure function?

ladymarlene, 2021-04-23:
Hi @all,

Thanks to Thomas for this nice project.

I have similar issue. Using Raspberry Pi 4 with ASI120MC-S the night time Capture doesn't start automatically at -6. GPS-Coordinates and time are set for my location.  

I tried to reinstall both Raspian and AllSky without success. I noticed the same behaviour described by hubivo: when I go to camera settings and just click "Save Changes" without making any changes, then the night time capture starts and stops in the morning at -6.

Any Ideas about that?