AllskyTeam/allsky #339: Supporting Arducam USB pi camera

spacetrace1982, 2021-03-16:
Hello 

Just wondered whether there is any support or any users who have used the Arducam’s USB camera family. Based on the 1/2.8” Sony IMX291 image sensor? https://thepihut.com/products/wide-angle-1080p-uvc-compliant-usb-camera-module-with-metal-case 
Any ideas on how and if I could get this working if I was to purchase one?

CleverJay, 2021-04-04:
I've ordered one of these cams as well. Mark McIntyre has an ALLSKY branch that supports IMX291 IP cams but not IMX291 USB Web cams.
 I'm going to try using VLC to create a stream, so that it appears as an IP cam. If that doesn't work, I will be begging for someone to add USB web cams to Capture.cpp

spacetrace1982, 2021-04-09:
I have the IP cam version too which I have now used for meteor detection on the Pi, but given the quality I'd have liked to have incorporated this ace set up from Thomas with the ability to use that cam which is just superb

CleverJay, 2021-04-13:
@spacetrace1982  I am using fswebcam and capturing jpegs from the USB camera. I am currently just using a cron job to save an image every minute to create a time-lapse. I don't know if there is a way to use fswebcam instead of raspistill (what the RaspiHQ cam uses) to capture images and control the camera in Allsky. I imagine that Thomas has neither the time nor the inclination to support more and more cameras.

thomasjacquin, 2021-04-14:
I might purchase one of those cameras at some point to test them. In the meantime if anyone was interested at taking a crack at it, you could modify the capture_RPiHQ.cpp file and more specifically this line https://github.com/thomasjacquin/allsky/blob/master/capture_RPiHQ.cpp#L156 (and any other mention of `raspistill`). I understand fswebcam probably doesn't have the same arguments as raspistill but that's a good starting point. And then recompile the code by running `make capture_RPiHQ`.

CleverJay, 2021-04-14:
Thanks @thomasjacquin  I'm capturing a time-lapse with the Arducam tonight, to get a baseline for settings and then will give swapping Raspistill calls a shot In capture_RPIHQ.cpp over the next few nights. Thanks again for Allsky!

CleverJay, 2021-04-15:
@spacetrace1982 @thomasjacquin  I've made some progress, in that I can run a capture script and control the settings of the camera using usbwebcam. Basically usbwebcam is using the v4l2 driver and you could also use VLC or player to capture using the same driver.  
Here is a controls list for the camera:
brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=32
saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=64
hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100
gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=100
power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=3 value=3
backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=156 value=156 flags=inactive
exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0

My nighttime capture script looks like this:
`#!/bin/bash

DATE=$(date +"%Y%m%d%H%M")

fswebcam -p MJPEG -r 1280x720 --no-banner --frames 300 set gain=50 set exposure_auto=0 set exposure-absolute=2500  /home/pi/allsky/webcamcap/$DATE.jpg
`
and then I am running a cron job to capture an image every minute.

The next step will be to attempt to replace raspistill with usbwebcam in capture_RPiHQ.cpp

CleverJay, 2021-04-15:
I replied all instances of raspistill with fswebcam in capture_RPiHQ.cpp.
when I try to compile I get the following error:
`pi@allsky:~/allsky $ make capture_RPiHQ
cc  capture_RPiHQ.cpp -o capture_RPiHQ -Wall -Wno-psabi -g -D_LIN -D_DEBUG    -lpthread  -DGLIBC_20  -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0
/usr/bin/ld: cannot find -lASICamera2
collect2: error: ld returned 1 exit status
make: *** [Makefile:60: capture_RPiHQ] Error 1
`
any ideas?

Zy0d0x0, 2021-06-13:
> I replaced all instances of raspistill with fswebcam in capture_RPiHQ.cpp.
> when I try to compile I get the following error:
> `pi@allsky:~/allsky $ make capture_RPiHQ cc capture_RPiHQ.cpp -o capture_RPiHQ -Wall -Wno-psabi -g -D_LIN -D_DEBUG -lpthread -DGLIBC_20 -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0 /usr/bin/ld: cannot find -lASICamera2 collect2: error: ld returned 1 exit status make: *** [Makefile:60: capture_RPiHQ] Error 1 `
> any ideas?

Hey not sure if you maanaged to fix this but using the following will help complile.

`
arm-linux-gnueabihf-g++  capture_RPiHQ.cpp -o capture_RPiHQ -Wall -Wno-psabi -g -D_LIN -D_DEBUG   -Llib/armv7 -I./include -lpthread  -DGLIBC_20 -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0
`

I have got the same issue with USB version of the IMX and have started editing the file up myself today.

Zy0d0x0, 2021-06-14:
This is still a working progresss but i have started intergrating the commands fswebcam supports into a seperate file then added so addition code to check for the camera. the lsusb part is where it fails to find other camera's.

```
pi@allsky:~/allsky $ ./allsky.sh
Making sure allsky.sh is not already running...
lsusb: option requires an argument -- 'D'
Usage: lsusb [options]...
List USB devices
  -v, --verbose
      Increase verbosity (show descriptors)
  -s [[bus]:][devnum]
      Show only devices with specified device and/or
      bus numbers (in decimal)
  -d vendor:[product]
      Show only devices with the specified vendor and
      product ID numbers (in hexadecimal)
  -D device
      Selects which device lsusb will examine
  -t, --tree
      Dump the physical USB device hierarchy as a tree
  -V, --version
      Show version of program
  -h, --help
      Show usage and help
Trying to automatically choose between cameras
USB MX291 camera found. Choosing MX291usb
Settings check done
CAMERA: MX291usb
CAMERA_SETTINGS: /home/pi/allsky/settings_MX291usb.json
Starting allsky camera...

 ******************************************
 *** Allsky Camera Software v0.6 | 2019 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ZWO ASI or RPi HQ camera

Add -h or -help for available options

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>
-Rob Musquetier

Capture Settings:
 Resolution: 0x0
 Exposure: 2500
 Auto Exposure: 0
 Gain: 50
 Brightness: 50
 Gamma: 50
 Delay: 10ms
 Daytime Delay: 30000ms
 Rotation: 0
 Flip Image: 0
 Filename: image.jpg
 Latitude: XXXXXN
 Longitude: XXXXXW
 Sun Elevation: 10
 Camera: USB MX291 camera

Determine if it is day or night using variables: desired sun declination angle: 10 degrees, latitude: XXXXN, longitude: XXXXXW
Check for day or night: NIGHT

Saving 2500 seconds exposure images with 10 ms delays in between...

Press Ctrl+Z to stop capturing images...

Capturing & saving image...
Capture command: nice fswebcam -d /dev/video0 -p MJPEG --greyscale -r 1920x1080 --set gamma=50 --set brightness=50 --set gain=50 --set exposure_auto=0 --set exposure_absolute=2500 image.jpg
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
No input was specified, using the first.
Setting Brightness to 50 (89%).
Setting Gamma to 50 (-5%).
Value is out of range. Setting anyway.
Setting Gain to 50 (50%).
--- Capturing frame...
Captured frame in 0.00 seconds.
--- Processing captured image...
Greyscaling image.
Writing JPEG image to 'image.jpg'.
Capturing & saving image done, now wait 0 seconds...
Determine if it is day or night using variables: desired sun declination angle: 10 degrees, latitude: XXXXXN, longitude: XXXXXW
Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
```

I am also getting images to the dashboard. I need to add couple other fixes to ensure the new image goes to guis live feed
other then that it is taking images using a custom config file for the  USB MX291 camera but this should also work for other cameras as it uses fswebcam the alternative to raspistill.

CleverJay, 2021-06-15:
For some reason My Arducam camera is only outputting analog YUV at 640X480 lately. The dev/video0 MotionJPEG is outputting garbage video, as is the H264 stream on dev/video3. So I am afraid I borked the camera. I am wondering whether to purchase another one in the hope that the results will be higher quality images than the RPi HQ camera.
Michael, can you post a night Image that you have captured with Arducam with your settings, please?
Thanks!

Zy0d0x0, 2021-06-17:
Hi Jay,

Sorry for the slow reply. I think the cameras are still not going to be
sensitive enough but I'm still messing around with the settings
to see if i can find a nice point.

CleverJay, 2021-06-20:
@Zy0d0x0  I am getting the same result in that, the camera isn't as sensitive as I would have thought. I am going to try frame stacking with  an app called Astro Dmx to capture with.

Blueshawk, 2021-09-25:
Has anyone made progress on this project? I also have a usb IMX291 I'd like to try with this terrific software setup.

Zy0d0x0, 2021-09-28:
I ended up moving to a raspberry pi camera in the end.

Blueshawk, 2021-10-01:
Bummer. I guess I should try to run some long exposure tests on mine and decide if I should return it while I can. Thanks for the heads up.

closed 2021-10-16 by EricClaeys