AllskyTeam/allsky #350: New RPiHQ camera not recognised...

Shand359, 2021-04-17:
I am trying to set up my allsky with a new RPi HQ camera. When I run to check the status of the camera nothing shows in the liveview window. If I run "sudo service allsky status" i get the following result:

Apr 17 14:58:12 allsky systemd[1]: Started All Sky Camera.
Apr 17 14:58:13 allsky allsky.sh[691]: RPiHQ Camera not found. Exiting.
Apr 17 14:58:13 allsky sudo[729]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop allsky
Apr 17 14:58:13 allsky sudo[729]: pam_unix(sudo:session): session opened for user root by (uid=0)
Apr 17 14:58:13 allsky systemd[1]: Stopping All Sky Camera...
Apr 17 14:58:13 allsky sudo[729]: pam_unix(sudo:session): session closed for user root
Apr 17 14:58:13 allsky systemd[1]: allsky.service: Main process exited, code=killed, status=15/TERM
Apr 17 14:58:14 allsky systemd[1]: allsky.service: Succeeded.
Apr 17 14:58:14 allsky systemd[1]: Stopped All Sky Camera.

Basically the camera is not found. 
I have since tried the camera on another Raspberry Pi and it works without issue....

thomasjacquin, 2021-04-17:
Is your camera enabled in the interfaces list?
https://projects.raspberrypi.org/en/projects/raspberry-pi-using/9

Shand359, 2021-04-21:
Not sure what’s happened, but it’s started working all of a sudden. Only issue I have now is the text above the images says “Limmen, Netherlands “ even although I have set my location in the settings

closed 2021-04-24 by Shand359