AllskyTeam/allsky #351: darkCapture invalid number error due to bash printf  localiation

agnunez, 2021-04-19:
darkCapture log an "invalid number" error continuosly, even with Dark Subtraction = false, due to reading of temperature.txt format expecting colon "." separated float in foreing buster installation in spanish ( comma "," separated float)

In theory, addling LC_NUMERIC="en_US.UTF-8" either in .bashrc or inline, in darkCapture:10  printf statement, should solve it, althout I did not manage to correct it yet.

This is a systenctl status log example:
 
allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-04-19 09:22:36 WEST; 5h 23min ago
 Main PID: 5889 (allsky.sh)
    Tasks: 7 (limit: 4915)
   CGroup: /system.slice/allsky.service
           ├─5889 /bin/bash /home/pi/allsky/allsky.sh
           └─5954 /home/pi/allsky/capture -angle -6 -autoexposure 1 -autogain 1 -bin 1 -brightness 50 -coolerEnabled 0 -darkframe 0 -daytimeDelay 60000 -delay 10 -exposure 1000 -extratext -extratextage 6

abr 19 14:32:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 47.7: número inválido
abr 19 14:33:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 48.5: número inválido
abr 19 14:34:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 47.7: número inválido
abr 19 14:35:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 47.7: número inválido
abr 19 14:36:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 47.7: número inválido
abr 19 14:37:09 elrito allsky.sh[5889]: /home/pi/allsky/scripts/darkCapture.sh: línea 10: printf: 48.5: número inválido

agnunez, 2021-04-25:
I opted by changing default location in raspbian to en_US.UTF8 and finally error disappeared.

closed 2021-04-25 by agnunez

mlincett, 2021-06-15:
Thanks @agnunez for suggesting a workaround but I think the issue should be reopened as long as the scripts are not locale-safe. I can confirm `printf` ignores the `LC_NUMERIC` setting if provided as a local variable, but using `export` seems to work.

Ideally one should set the proper environment for the service. Something of the kind:

```
[Service]
...
Environment="LC_NUMERIC="en_US.UTF-8"
```
in the `allsky.service` file should do the trick.

I can test it and provide a pull request.

The problem will persist if the script is called manually, but it still seems more logic than doing an `export` in `darkCapture.sh`.

Opinions?