AllskyTeam/allsky #352: time format

rabit9, 2021-04-20:
Hello, super nice Programm. One question, how can i change the time format in top of the Picture from American to Europe format. Exemple 04/20/21 to 20/04/21

thomasjacquin, 2021-04-20:
Change the format on this line https://github.com/thomasjacquin/allsky/blob/master/capture.cpp#L93
and run `make capture` to recompile.

rabit9, 2021-04-20:
I made changes in line 93, then in the allsky folder I did sudo make capure and restarted raspberry, but nothing has changed

thomasjacquin, 2021-04-20:
Did you run `make capture` to recompile the program?

rabit9, 2021-04-20:
I did sudo make capure

thomasjacquin, 2021-04-20:
Ok, there's no need for sudo when compiling the program but that shouldn't change anything. 
Are you using a ZWO camera or a RPiHQ camera?

rabit9, 2021-04-20:
a RPiHQ camera

thomasjacquin, 2021-04-20:
In that case, the date is the output of the raspistill `-annotate` option and I don't think we have control over that.

AndreasLMeg, 2021-06-07:
https://www.raspberrypi.org/documentation/raspbian/applications/camera.md
https://man7.org/linux/man-pages/man3/strftime.3.html

`-a 8 -a "ABCD %Y-%m-%d %X %Z"` works for me

EricClaeys, 2021-06-13:
I modified capture.cpp to accept a date/time format, and modified the admin interface to prompt for the format.  Works very well; you can then change the format without having to update the .cpp file.
How can I propose these changes become part of the official software?

thomasjacquin, 2021-06-13:
Hi Eric, you can create a Pull Request and I'll review the code and merge it into the master branch. Thanks

EricClaeys, 2021-06-13:
Thomas,
I have a bunch of bug fixes and enhancements. Should I put them all in the file in the pull request, or do separate requests for each feature?  I also put a comment with my initials (ECC) so it would be obvious what I changed.  Is that necessary?

Question:  the getTime() function checks if the seconds are the same as the last call, and if so, returns 0.  What's the reason for that?  getTime() is only called once, and one of my changes includes calling it a second time, so I had to create a version of getTime() that didn't do the check.

Eric

ckuethe, 2021-10-05:
capture now includes a `-timeformat` flag which can be used to format time in an arbitrary format.

closed 2021-10-05 by ckuethe