AllskyTeam/allsky #353: AllSky website file format

KaktuMAN, 2021-04-21:
Hello, 
I have the AllSky website on the same Raspberry PI as the AllSky software.
I can access to the videos but when I want to check startrails, they are not appearing on the website.
(And how do we configure the live view ?)

thomasjacquin, 2021-04-21:
Make sure startrails are created at `/home/pi/allsky/images/20210420/keogram/keogram-20210420.jpg`.
If the keogram isn't created properly, it won't be transferred to the Allsky website.

KaktuMAN, 2021-04-21:
Hi, all the generated files are correctly transferred but they aren't visible in the website

thomasjacquin, 2021-04-21:
Ok try these:
```
sudo apt-get install php-gd
cd /var/www/html
sudo chown -R pi:www-data allsky
sudo chmod -R 775 allsky
```

KaktuMAN, 2021-04-22:
Still not working. In videos I have the timelapse he is visible in the timelapse page (on the web). In keograms I have the keogram but the keogram isn't visible in the keogram page.

closed 2021-04-28 by KaktuMAN