AllskyTeam/allsky #356: Location stamp....

Shand359, 2021-04-24:
I’ve tried numerous times to get the location and time zone to change for the camera location but it keep switching back to Limmen, Netherlands.....

uduluiz, 2021-05-27:
thanks for the feedback. One more doubt, the lens that comes with the camera has how many degrees?

PhotonPup, 2021-07-24:
For me, though I edited the config.js file none of my edits were able to display so I edited the index.htl file directly. Row 42-47 you can set entried directly and they will not change.

EricClaeys, 2021-08-01:
Often when changing he config.js file, you need to force your browser to re-read the file.  You can either clear the cache, or press F12 when on the page, and in the "Network" tab, check the option to disable the cache, then refresh the page.

Shand359, where are you changing the information?  And exactly what are you changing, and where?  When you say it keeps "switching back", does that mean it's correct once, then not again, or it's never correct?  If the latter, please try the above fix.

Eric

closed 2021-10-16 by EricClaeys