AllskyTeam/allsky #357: Angle option change brightness

nacho3g, 2021-04-24:
I am using a raspberry pi 4 and a HQ Cam, when I change the "angle" option in Camera Settings, it totally changes the brightness of the image.
I give an example changing only angle to -6 and +6.
I do not understand where the problem is since this option should not affect the quality of the image.
Thanks for the help.

[image]
[image]

EricClaeys, 2021-10-16:
Are both images taken at the same time, give or take a few seconds?  And are they taken during "daytime" or "nighttime"?

EricClaeys, 2022-05-31:
Closing Issue - no response from user in over 6 months.  The day versus night settings are likely causing the problem.  When changing the angle, it's likely the nighttime settings are used during the day, severely over exposing the image.  And at night, if daytime exposure is used the image will be very dark.

closed 2022-05-31 by EricClaeys