AllskyTeam/allsky #358: Automatic gain transition

loloderu, 2021-04-25:
Hi,

When there is a transition between night and day the gain goes to 0 but also the exposure.
In order to have a smoother transition would it be possible to start with the exposure to the last exposure made at night ?

Regards

EricClaeys, 2021-06-13:
It still wouldn't look very good for a couple frames if you had nighttime gain set to anything > 0.
For example, going from gain 200, exposure 60 seconds to gain 0, 60 seconds would produce a significantly dark image.  I suppose it would be better than going to gain 0 exposure 32 micro seconds, which is the starting point for daytime exposures.

AndreasLMeg, 2021-06-17:
Hi,

I try to solve this transition problem for my RaspberryPi HQ camera [https://github.com/AndreasLMeg/allsky.git]
Here for example is the video from last night: [https://youtu.be/wPBJr92lSeI]

loloderu, 2021-06-17:
Hi Andreas, i See you have a modified version of thomasjacquin allsky software.
From my part  have gone also finally to a Raspberry Pi HQ but used instead o fraspitill , raspiraw to have longer than 21s exposure and no high delay between exposure ( 2 or 3 time the pose with raspistill).
Any way cold you explai what are the impovement you made?

AndreasLMeg, 2021-06-17:
In burst mode it's possible to use shutterspeeds up to ~200s, I have set the limit to 60s

Using the following call to generate an image: (from DAYLIGHT to NIGHT)
raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 1 --analoggain 1 --exif IFD0.Artist=li_1_1 --awb auto
...
raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 8000000 --analoggain 1 --exif IFD0.Artist=li_8000000_1 --awb auto
...
raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 1 --exif IFD0.Artist=li_60000000_1 --awb auto
raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 2 --exif IFD0.Artist=li_60000000_2 --awb auto
...
raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 16 --exif IFD0.Artist=li_60000000_16 --awb auto

Here is a short description: 
* Simple algorithm - the shutter speed and gain are adjusted based on the averaged exposure value
* mode mean tries to make well exposed images
* underexposed: image < (mean value - threshold) -> increase shutter time or gain, overexposed: image > (mean value + threshold) -> decrease shutter time or gain
* For step ...-2, -1, 0, 1, 2, ... -> shuttertime=2^(step/shuttersteps)
* shuttersteps=1: shuttertime 1s, 2s, 4s, 8s,... 
* shuttersteps=3: 1s, 1,26s, 1,59s, 2s,...

more: https://github.com/AndreasLMeg/allsky/blob/master/README%20mode%20mean.md

So I get "well-exposed" images the hole day - But only the night is important for me. 
Impovement:
1. no fixed gain the hole night (Problem with sunset, sunrise and moon)
2. Timelapse is more like nature (blue hour to dark night and back to blue hour)

I tried to implement a "mode", so the original software should still work.

loloderu, 2021-06-18:
Ok I though Raspistill was limited to 21s exposure, but I see that it was updated and that if we ask for 60s, the delay is 60s: that's great.
Thanks for your version with the mode mean! this sound very good and I will test it.

AndreasLMeg, 2021-06-20:
This was last night: https://youtu.be/4RBActnQPds
Now shuttertime, gain and WB infos are shown. So you can see the "mean mode" working

NeilBuc007, 2021-06-23:
> This was last night: https://youtu.be/4RBActnQPds
> Now shuttertime, gain and WB infos are shown. So you can see the "mean mode" working

I am testing this out with one of my RPiHQ cams. Well done. Looks ok so far.
https://stenton.allsky-scotastro.co.uk
Will see what happens this evening when it gets dark.
Thanks for effort.

AndreasLMeg, 2021-06-23:
Thx for testing.
my setting for the last nights, but I'm still unsure:
MODE=1 # mean
# Mode 1: mean - Simple algorithm - the shutter speed and gain are adjusted based on the averaged exposure value
MEAN_VALUE=0.3
MEAN_THRESHOLD=0.05
MEAN_SHUTTERSTEPS=6

I was even able to take some photos of thunderstorms with lightning
Good luck !

[image]

NeilBuc007, 2021-06-24:
> This was last night: https://youtu.be/4RBActnQPds
> Now shuttertime, gain and WB infos are shown. So you can see the "mean mode" working

My first test did not work out as well as as your video. I suspect it must be settings I have wrong. Could you share your config.sh and settings_RPiHQ.json?

Many thanks.

NeilBuc007, 2021-06-24:
Thanks. That is a cool capture. I had:
MEAN_VALUE=0.5
MEAN_THRESHOLD=0.05
MEAN_SHUTTERSTEPS=3

So will change to what you suggested and try.

AndreasLMeg, 2021-06-24:
MEAN_SHUTTERSTEPS=6 will reduce the flicker

angle to 6° to get more of sunset/sunrise

max main to 14

Auto WB: OFF

 

Can you send me some lines from your logging tonight. So I can check the parameters.

 

 

 

Gesendet: Donnerstag, 24. Juni 2021 um 12:35 Uhr

NeilBuc007, 2021-06-24:
Will do. 

I had angle to 6 degrees but have actually set it to 3 degrees for tonight to capture more.

I did translate the Exposure Level, Exposure Time and Reinforcement  words that are printed out to the log.
You can see it swings a bit either side, but eventually it settles.

http://stenton.allsky-scotastro.co.uk/videos/allsky-20210623.mp4

Also It is actually v0.7 as its a fresh install. (needs to be updated that)

Cheers

Neil

p.s. I can send you the proper full log file if needed.

Some logs from last night:
Jun 23 23:16:57 allsky allsky.sh[395]: Starting allsky camera...
Jun 23 23:16:58 allsky allsky.sh[395]: mode mean
Jun 23 23:16:59 allsky allsky.sh[395]: #033[32m ******************************************
Jun 23 23:16:59 allsky allsky.sh[395]: #033[32m *** Allsky Camera Software v0.6 | 2019 ***
Jun 23 23:16:59 allsky allsky.sh[395]: #033[32m ******************************************
Jun 23 23:16:59 allsky allsky.sh[395]: #033[32mCapture images of the sky with a Raspberry Pi and an ZWO ASI or RPi HQ camera
Jun 23 23:16:59 allsky allsky.sh[395]: #033[33mAdd -h or -help for available options
Jun 23 23:16:59 allsky allsky.sh[395]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Jun 23 23:16:59 allsky allsky.sh[395]: #033[0mContributors:
Jun 23 23:16:59 allsky allsky.sh[395]: -Knut Olav Klo
Jun 23 23:16:59 allsky allsky.sh[395]: -Daniel Johnsen
Jun 23 23:16:59 allsky allsky.sh[395]: -Yang and Sam from ZWO
Jun 23 23:16:59 allsky allsky.sh[395]: -Robert Wagner
Jun 23 23:16:59 allsky allsky.sh[395]: -Michael J. Kidd - <linuxkidd@gmail.com>
Jun 23 23:16:59 allsky allsky.sh[395]: -Rob Musquetier
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: /home/pi/allsky/capture_RPiHQ: -angle
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -angle: -6
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -autoexposure: 1
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -autogain: 1
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -awb: 0
Jun 23 23:16:59 allsky allsky.sh[395]: argument: -awb: 0 not use !
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -background: 0
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -bin: 2
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -brightness: 50
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -daytimeDelay: 60000
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -delay: 5000
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -exposure: 30000
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -filename: image.jpg
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -flip: 2
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -fontcolor: 255
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -fontsize: 32
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -gain: 4
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -gamma: 50
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -height: 0
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -latitude: 55.96N
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -longitude: 2.606W
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -quality: 95
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -rotation: 0
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -showDetails: 0
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -time: 1
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -wbb: 1.47
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -wbr: 3.24
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -width: 0
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -mode: 1
Jun 23 23:16:59 allsky allsky.sh[395]: mode: 1
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -mean-value: 0.5
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -mean-threshold: 0.05
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -mean-shuttersteps: 3
Jun 23 23:16:59 allsky allsky.sh[395]: Processing argument: -daytime: 1
Jun 23 23:16:59 allsky allsky.sh[395]: #033[0m- Resolution: 4096x3040
Jun 23 23:16:59 allsky allsky.sh[395]: - Pixel Size: 1.55μm
Jun 23 23:16:59 allsky allsky.sh[395]: - Supported Bin: 1x, 2x and 3x
[... 178 lines cut ...]
Jun 23 23:19:18 allsky allsky.sh[395]: asiExposure: 30000000
Jun 23 23:19:18 allsky allsky.sh[395]: asiGain: 4.0000
Jun 23 23:19:18 allsky allsky.sh[395]: mean_shuttersteps: 3.0000
Jun 23 23:19:18 allsky allsky.sh[395]: Mean: 0.5023 Exposure Level:0 Exposure TIme:1.00000000 Reinforcement:1
Jun 23 23:19:18 allsky allsky.sh[395]: Capturing & saving image done, now wait 5 seconds...
Jun 23 23:19:18 allsky allsky.sh[395]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: 55.96N, longitude: 2.606W
Jun 23 23:19:18 allsky allsky.sh[395]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Jun 23 23:19:18 allsky allsky.sh[395]: Capturing & saving image...
Jun 23 23:19:18 allsky allsky.sh[395]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 2  --width 2028 --height 1520 --exposure off --shutter 1000000 --analoggain 1 --exif IFD0.Artist=li_1000000_1 --awb off --awbgains 3.24,1.47 --vflip --saturation 50 --quality 95 -a 1036 -ae 32,0xff,0x808000
Jun 23 23:19:18 allsky allsky.sh[395]: image.jpg 0.576945
[... cut ...]
Jun 23 23:19:31 allsky allsky.sh[395]: Capturing & saving image...
Jun 23 23:19:31 allsky allsky.sh[395]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 2  --width 2028 --height 1520 --exposure off --shutter 1000000 --analoggain 1 --exif IFD0.Artist=li_1000000_1 --awb off --awbgains 3.24,1.47 --vflip --saturation 50 --quality 95 -a 1036 -ae 32,0xff,0x808000
Jun 23 23:19:31 allsky allsky.sh[395]: image.jpg 0.574823
Jun 23 23:19:31 allsky allsky.sh[395]: Resizing
Jun 23 23:19:32 allsky allsky.sh[395]: Uploading

AndreasLMeg, 2021-06-24:
[settings_RPiHQ.zip](https://github.com/thomasjacquin/allsky/files/6710447/settings_RPiHQ.zip)

sorry for the wrong information. I use angle -6° 
there is a "fast forward" mode, maybe only working for mean_shuttersteps=6 (flicker/swinging)

And one line of logging: 
Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 6 --exif IFD0.Artist=li_60000000_6 --awb off --awbgains 2,1.6 --hflip --vflip --saturation 50 --quality 100 -a 1036 -a "Meggenhofen (li) 60000000 6 2 1.6" -ae 32,0xff,0x808000

I will merge your translation

AndreasLMeg, 2021-06-24:
I'm using mode 3 (binning 1x1) because in mode 2 (binning) the camera or driver take use the parameter shutter.
my results with mode 3 are better ;-)

NeilBuc007, 2021-06-24:
> I'm using mode 3 (binning 1x1) because in mode 2 (binning) the camera or driver take use the parameter shutter.
> my results with mode 3 are better ;-)

Thanks Andreas. I will switch back to Bin 1. I can always reduce the image size before creating the timelapse. It might be fine as it is a RPI4b with 4GB so it should create the MP4 file.

AndreasLMeg, 2021-06-24:
translation and default values: done
https://github.com/AndreasLMeg/allsky/commit/9535fa7ecd858228f734a721caf84d35bfc41321

AndreasLMeg, 2021-06-25:
last night: https://youtu.be/QtHYKL4hoH0

AndreasLMeg, 2021-06-26:
Hi Neil,

is it possible to clone your allsky site and deliver it with my allsky data? (via ftp)
I don't have my own webspace and don't want to do this to myself.
Best regards Andreas

rodrigoluk, 2021-07-17:
Hi,
I'm getting a lot of red glow at night and few stars and no milky way. I'm using Andreas auto gain on Raspberry Pi HQ camera. Using a Pi4. Gain tops at 16 all night. Any advise?

Thanks,

R.

AndreasLMeg, 2021-07-17:
mean mode is still under construction...

my settings (sorry I'm "color blind" !)
[image]
[image]

and the result
[image]

rodrigoluk, 2021-07-17:
Hi Andreas, - Thanks for your reply! I will try the settings and continue to experiment. :-)

bbillp, 2021-07-17:
Even after making a new darks folder and rebooting why is the dark image sequence number starting at 38.jpg then at 39.jpg it stops incrementing and just replaces 39.jpg file every few minutes?
Bill Prats

EricClaeys, 2021-07-17:
Bill,
Dark image file names are the temperature, so 38.jpg is a dark taken with a sensor temp of 38 degrees.

NeilBuc007, 2021-07-19:
[image]

Try using  these settings too:
[image]

AndreasLMeg, 2021-07-19:
my test masking out the horizon from last night. 
[image]

Dentonknifeworks, 2021-07-21:
Well I have now joined in trying the Automatic Gain Transition.
Setup it today and will try it for a few weeks. I hope it works I have one all sky in the back yard that is been working for the last year. I now have built a new one for my RV for when we go to darker Skys and I can use this new one. I have had it out two times and finding that I spend a lot of time adjusting and testing in the darker Skys only to find in the morning I should have used more light gathering. So I am hoping that this automatic gain will help in getting a close to good video in the darker Skys.

Dentonknifeworks, 2021-07-22:
I am having a issue that I don't think is related to this. The pi locks up after 15 or so images at night. I find the processor is at 100% and takes a hard reboot to fix. I seem to remember this issue and was able to fix it but can't find it now. Thinking it had to due with the image size needed to be reduced. Anyone had this issue?

rodrigoluk, 2021-07-22:
I had the same issue and a firmware update resolved it. A new raspistill
was installed.
Note that this will depend on your camera model, version, etc.

Try doing a full update and then a "sudo apt full-upgrade".

Good luck!

R.

AndreasLMeg, 2021-07-22:
https://youtu.be/jBPuhz8ju6A

Downgrade- read description of video
Works since 2 months

Dentonknifeworks, 2021-07-23:
I did the upgraded of the firmware and no issues last night. This is working well photo attached the one on the left is the one I have been running the last year. The one on the right in next to it at the same time and you can see that its not over exposed. Liking the results so far.
[image]