AllskyTeam/allsky #359: Preview not working

EricClaeys, 2021-04-25:
I have a ZWO 178MC and when "Preview" is set to "1", the allsky program gives an error:
 **Unable to init server: Could not connect: Connection refused**

It fails in the Display() function at the call to cvShowImage("video", pImg);
All calls to ASIGetVideoData() also fail with return code 13 (buffer too small), pRgb.step[0]=9288, and pRgb.rows=0.

If I start the program by ./allsky.sh I get this error:
**OpenCV Error: Bad flag (parameter or structure field) (Unrecognized or unsupported array type) in cvGetMat, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp, line 2501
		terminate called after throwing an instance of 'cv::Exception'
 		 what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp:2501: error: (-206) Unrecognized or unsupported array type in function cvGetMat**

Any ideas?

laheller, 2021-07-10:
@EricClaeys 

I have the same problem :-/

bbillp, 2021-07-10:
Where do these messages come from because I don’t see my posts.  Bp

EricClaeys, 2021-07-10:
Laheller,
I was never able to figure this out.  The preview feature works in the ZWO "demo" program that comes with their SDK, although its code is somewhat different.  I don't know enough about the CV code to be able to debug it.

Eric

EricClaeys, 2021-07-10:
Bbillp,
I see your post asking where the messages come from.  I assume at some point you responded to a message on this thread, or subscribed to it, so GitHub is now sending you emails when there is an update.

bbillp, 2021-07-10:
Is this mail list a substitute for posts on allsky at github because my posts there don’t come up in these emails. Today are my first posts in a couple of years so i missed a lot of info.  
Bill

EricClaeys, 2021-10-16:
Closing issue.  preview doesn't work over VNC viewer.  It DOES work when logged in directly to the Pi.

closed 2021-10-16 by EricClaeys

laheller, 2021-10-18:
> Closing issue. preview doesn't work over VNC viewer. It DOES work when logged in directly to the Pi.

@EricClaeys 
No, it does NOT. At least on my RPI it throws the same error as above even when logged in directly.

linuxkidd, 2021-10-24:
This is likely related to PR #507 
> NOTE: I had quite a bit of issue getting the Display() function to compile, but finally got something workable -- although, I've not tested the Display capability. I would love for someone to test the Display code prior to merge. Otherwise, the rest of the code compiles fine on both Raspberry Pi OS based on Debian Buster with OpenCV 3.2.0, as well as Debian Bullseye with OpenCV 4.5.1.