AllskyTeam/allsky #360: First 3 images are identical

EricClaeys, 2021-04-25:
When starting allsky, the first 3 images are identical, although the overlay information changes.  I know the images are identical because I have the ZWO ASI178MC camera pointed at a clock and the time doesn't change until the 4th picture.

I've tried changing the exposure, auto exposure, gain, and auto gain but it didn't help.

Looking at the code in capture.cpp, I can't see anything that would buffer the images.

Any ideas?

Astro-Pete, 2021-04-27:
I just checked on my daytime capture folder and can confirm this. The clouds don’t move for the first 3 images.

EricClaeys, 2021-04-30:
As a workaround, I'm going to change capture.cpp to take 3 very short exposure images and not save them, then go into the loop taking normal images.

EricClaeys, 2021-06-13:
The workaround to take 3 short exposures and not save them works very well.  How do I propose the changes be incorporated into the official software?

EricClaeys, 2021-09-16:
Fixed in version 0.8.

closed 2021-09-16 by EricClaeys