AllskyTeam/allsky #364: IMX290

Kilis71, 2021-04-27:
I would have an IMX290 cmos is it possible to use it, is it supported?

scorot, 2021-05-10:
Not tested myself, but as long as you can capture images with raspistill with the this sensor, this should work.

KarstenGruettner, 2023-05-20:
As a note, **it doesn't work.**

During the installation, the command 
libcamera-still -cc_file '/home/pi/allsky/config/cc.json
brings no result. This is how it fails.

EricClaeys, 2023-05-20:
@KarstenGruettner is this a ZWO camera or a Raspberry Pi (or compatible) camera?
Do you have a URL that lists the specifications for the camera?

EricClaeys, 2023-05-20:
@KarstenGruettner please ATTACH the file `~/allsky/config/installation_logs/install.sh.log`

KarstenGruettner, 2023-05-20:
> @KarstenGruettner is this a ZWO camera or a Raspberry Pi (or compatible) camera? Do you have a URL that lists the specifications for the camera?

Hello Eric,

Thank you for your quick response.
It is compatible with Raspberry pi and connected to CSI. I can also use it with libcamera via the command line.

KarstenGruettner, 2023-05-20:
> @KarstenGruettner please ATTACH the file `~/allsky/config/installation_logs/install.sh.log`

[install.sh.log](https://github.com/thomas
[image]
jacquin/allsky/files/11523037/install.sh.log)

I couldn't find anything in the error log. I am then in the single steps of the install.sh

KarstenGruettner, 2023-05-20:
[install.sh.log](https://github.com/thomasjacquin/allsky/files/11523067/install.sh.log)

EricClaeys, 2023-05-20:
Interesting, the installation thinks it's a RPi HQ camera, but it's not so that's a bug I will need to look into.

Please attach the `~/allsky/config/cc.json` file.
Also, run `vcgencmd get_camera`.

KarstenGruettner, 2023-05-21:
> Interesting, the installation thinks it's a RPi HQ camera, but it's not so that's a bug I will need to look into.
> 
> Please attach the `~/allsky/config/cc.json` file. Also, run `vcgencmd get_camera`.

Thanks a lot for your help.

~/allsky/config/cc.json does not exist. 

i debugged the source (i am developer too , I'm not perfect in python, shell scripting, but I can do the basics )
As I wrote   libcamera-still -cc_file '/home/pi/allsky/config/cc.json
does not create a file. I think that's the problem here. I found no documentation about the cc_file - parameter.

I tried vcgencmd get_camera and libcamera-still --list-cameras

[image]

EricClaeys, 2023-05-21:
Thanks for that info.  We should be able to support it.  After I make the changes I will need your help to test them.

`libcamera-still` doesn't have a `cc_file` option.  The Allsky program `capture_RPi` does.  Please run
~~~
cd ~/allsky
./install.sh --debug --function recreate_options_file
~~~
and then ATTACH `~/allsky/config/installation_logs/install.sh.log`
Also, please run `ls -l config tmp`

KarstenGruettner, 2023-05-21:
> Thanks for that info. We should be able to support it. After I make the changes I will need your help to test them.
> 
> `libcamera-still` doesn't have a `cc_file` option. The Allsky program `capture_RPi` does. Please run

that explains a lot ;-) 

> ```
> cd ~/allsky
> ./install.sh --debug --function recreate_options_file
> ```
> 
> and then ATTACH `~/allsky/config/installation_logs/install.sh.log` Also, please run `ls -l config tmp`

I deleted "~/allsky/config/installation_logs/install.sh.log" first and then called the command "./install.sh --debug --function recreate_options_file". No new log file was created

[image]

EricClaeys, 2023-05-21:
Thanks.  Clearly a bug when it can't find the camera.  Look at the 4th line in your screenshot. It should say `New=[RPi]`, not `New=[       #Updated during installation]`

KarstenGruettner, 2023-05-21:
Thanks you very much.
I tried to fake the camera  and set in install.sh line 319 CAMERA_TYPE="RPi", but it can not build the json-files (or later use in createAllyskyOptions.php)

[image]

I'm afraid it's not that easy ;-)

EricClaeys, 2023-05-22:
@KarstenGruettner   No, Allsky needs to know the specifications for the camera, which you provided via the `libcamera-still --list-cameras` output.  I need to put that in Allsky.  Also need to fix the bug.  You should have gotten a message saying something like "imx290 cameras are not currently supported by Allsky...".

KarstenGruettner, 2023-05-30:
First light with imx290

[image]

KarstenGruettner, 2023-05-30:
Sorry I couldn't wait ;-)
I don't want to change their source code without consultation. 
in ASI_functions.cpp
I add the two arrays

	{ "imx290", 0, "IMX290 CSI Board", 0, 1920, 1080, ASI_TRUE,
		BAYER_RG, {1, 2, 0}, {ASI_IMG_RGB24, ASI_IMG_END}, 2.9, ASI_FALSE,
		12, ASI_FALSE, ASI_FALSE}, // nur zum Testen

	{ // imx290
		// TODO: Update as necessary
		{ "Gain", "Gain", 16.0, 1, 1, NOT_SET, ASI_TRUE, ASI_TRUE, ASI_GAIN },
		{ "Exposure", "Exposure Time (us)", 200 * US_IN_SEC, 1, 10000, NOT_SET, ASI_TRUE, ASI_TRUE, ASI_EXPOSURE },
		{ "WB_R", "White balance: Red component", 10.0, 0.1, 2.5, NOT_SET, ASI_TRUE, ASI_TRUE, ASI_WB_R },
		{ "WB_B", "White balance: Blue component", 10.0, 0.1, 2.0, NOT_SET, ASI_TRUE, ASI_TRUE, ASI_WB_B },
		{ "Temperature", "Sensor Temperature", 80, -20, NOT_SET, NOT_SET, ASI_FALSE, ASI_FALSE, ASI_TEMPERATURE },
		{ "Flip", "Flip: 0->None, 1->Horiz, 2->Vert, 3->Both", 3, 0, 0, NOT_SET, ASI_FALSE, ASI_TRUE, ASI_FLIP },
		{ "AutoExpMaxGain", "Auto exposure maximum gain value", 16.0, 1, 16.0, NOT_SET, ASI_FALSE, ASI_TRUE, ASI_AUTO_MAX_GAIN },
		{ "AutoExpMaxExpMS", "Auto exposure maximum exposure value (ms)", 200 * MS_IN_SEC, 1, 60 * MS_IN_SEC, NOT_SET, ASI_FALSE, ASI_TRUE, ASI_AUTO_MAX_EXP },
		{ "ExposureCompensation", "Exposure Compensation", 10.0, -10.0, 0, NOT_SET, ASI_FALSE, ASI_TRUE, EV },
		{ "Brightness", "Brightness", 1.0, -1.0, 0, NOT_SET, ASI_FALSE, ASI_TRUE, ASI_AUTO_TARGET_BRIGHTNESS },
		{ "Saturation", "Saturation", 15.99, 0.0, 1.0, NOT_SET, ASI_FALSE, ASI_TRUE, SATURATION },
		{ "Contrast", "Contrast", 15.99, 0.0, 1.0, NOT_SET, ASI_FALSE, ASI_TRUE, CONTRAST },
		{ "Sharpness", "Sharpness", 15.99, 0.0, 1.0, NOT_SET, ASI_FALSE, ASI_TRUE, SHARPNESS },

		{ "End", "End", 0.0, 0.0, 0.0, 0.0, ASI_FALSE, ASI_FALSE, CONTROL_TYPE_END },
	},
	{ // arducam_64mp, raspistill.  Not supported.
		{ "End", "End", 0.0, 0.0, 0.0, 0.0, ASI_FALSE, ASI_FALSE, CONTROL_TYPE_END },
	},

EricClaeys, 2023-05-30:
@KarstenGruettner   That's the same thing I did but I used a generic name of "imx290 60.00 fps".  With the RPi-compatible cameras there could be multiple vendors using the same chip.

Support for the IMX290 will officially be in the point release in a week or so, so I'm closing this Issue.  Thanks for testing it.

closed 2023-05-30 by EricClaeys

KarstenGruettner, 2023-05-31:
I'm glad to read that. Thank you for the quick implementation.