AllskyTeam/allsky #365: limit live-image view to max 100% of size

Lorenzi70, 2021-04-29:
I have a problem with the view of the live image, as it is always scaled to 100% of the screen and so I can never see it complete even when I use a large screen. in Fact it is enlarged on high res screens.
Is it possible to keep down scaling as today the image if watched on small screens (e.g. a phone) but limit the upper scaling to 100% if a high res monitor is used?
I tried to remove the "style=100%" in liveview.php but it doesn't really work (in fact it works for few seconds than it again scale it up).
Thanks
Marco

thomasjacquin, 2021-05-03:
Try this.

```
<div id="live_container" style="background-color: black; margin-bottom: 15px; height: calc(100vh - 95px)">
   <img id="current" class="current" src="current/liveview-<?php echo $camera_settings_array['filename'] ?>" style="height:100%">
</div>
```

Lorenzi70, 2021-05-03:
Thanks Thomas. 
Unfortunately it works only until the screen is automatically refreshed (for the first 15/20 seconds), than the image again stretches to the full frame width. Same if I click on the image and than click esc to the full window, the image will stretch to the full frame width..

jcauthen78, 2021-05-10:
I can second this problem on both Firefox & Chrome (running MacOS). Atemped the change stated above, the image is correct until the next image capture & reload. I even resized the image in the captureNight.sh file, but the browser window stretches the image back up and out of the frame as if it wasn't resized - inspecting the displayed image, the resize is successful, and saves as the resized size, but neither browser reflect the change.

ayladehond, 2021-05-13:
Also change this in index.php ,rules 153 and 155
.css("height", "100%")
                .on('load', function () {
                    if (!this.complete || typeof this.naturalHeight == "undefined" || this.naturalHeight == 0)

And in public.php, rules 51 and 53

.css("height:100%")
                                .on('load', function() {
                                    if (!this.complete || typeof this.naturalHeight == "undefined" || this.naturalHeight == 0)

Now you should be fine.

closed 2021-08-20 by Lorenzi70