AllskyTeam/allsky #366: no keogram, startrails, timelapse

paulfindus, 2021-05-03:
Hi folks,
after a new installation of raspbian and allsky, files for keogram, startrails and timelapse dont appear anymore.
Yes, the switches are set to "true".
Yes the "chown pi:pi -R allsky" is run.
Yes the firmware of ZWO 120Mc is updated.
The messages from the on are the same for "ever".
Especially in line 10 the exposure seems to be interrupted..."...value: 1712 mGenerating Keogram"?
Thank you for any help...
From Peter

1 May  2 04:45:06 allsky allsky.sh[587]: Sleeping: 18133 ms
2 May  2 04:45:06 allsky allsky.sh[587]: Exposure value: 1793086 µs
3 May  2 04:45:06 allsky allsky.sh[587]: 20210502 04:44:48
4 May  2 04:45:06 allsky allsky.sh[587]: Auto Exposure value: 1793 ms
5 May  2 04:45:06 allsky allsky.sh[587]: Sleeping: 18217 ms
6 May  2 04:45:06 allsky allsky.sh[587]: Exposure value: 1712398 µs
7 May  2 04:45:06 allsky allsky.sh[587]: 20210502 04:45:06
9 May  2 04:58:40 allsky allsky.sh[587]: Auto Exposure value: 1712 mGenerating Keogram
10 May  2 04:58:44 allsky allsky.sh[587]: /home/pi/allsky/allsky.sh: Zeile 86:   672 Speicherzugriffsfehler  $ALLSKY_HOME/capture $ARGUMENTS
May  2 04:58:49 allsky allsky.sh[2041]: Making sure allsky.sh is not already running...
May  2 04:58:49 allsky allsky.sh[2041]: can't get debug descriptor: Resource temporarily unavailable
May  2 04:58:49 allsky allsky.sh[2041]: Trying to automatically choose between ZWO and RPI camera
May  2 04:58:49 allsky allsky.sh[2041]: ZWO camera found. Choosing ZWO
May  2 04:58:49 allsky allsky.sh[2041]: Settings check done
May  2 04:58:49 allsky allsky.sh[2041]: CAMERA: ZWO
May  2 04:58:49 allsky allsky.sh[2041]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
May  2 04:58:49 allsky allsky.sh[2041]: Starting allsky camera...
May  2 04:58:59 allsky allsky.sh[2041]: /home/pi/allsky/allsky.sh: Zeile 86:  2104 Speicherzugriffsfehler  $ALLSKY_HOME/capture $ARGUMENTS
May  2 04:59:05 allsky allsky.sh[2113]: Making sure allsky.sh is not already running...
May  2 04:59:05 allsky allsky.sh[2113]: can't get debug descriptor: Resource temporarily unavailable
May  2 04:59:05 allsky allsky.sh[2113]: Trying to automatically choose between ZWO and RPI camera
May  2 04:59:05 allsky allsky.sh[2113]: ZWO camera found. Choosing ZWO
May  2 04:59:05 allsky allsky.sh[2113]: Settings check done
May  2 04:59:05 allsky allsky.sh[2113]: CAMERA: ZWO
May  2 04:59:05 allsky allsky.sh[2113]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
May  2 04:59:05 allsky allsky.sh[2113]: Starting allsky camera...
May  2 04:59:15 allsky allsky.sh[2113]: /home/pi/allsky/allsky.sh: Zeile 86:  2176 Speicherzugriffsfehler  $ALLSKY_HOME/capture $ARGUMENTS
May  2 04:59:20 allsky allsky.sh[2185]: Making sure allsky.sh is not already running...
May  2 04:59:20 allsky allsky.sh[2185]: can't get debug descriptor: Resource temporarily unavailable
May  2 04:59:20 allsky allsky.sh[2185]: Trying to automatically choose between ZWO and RPI camera
May  2 04:59:20 allsky allsky.sh[2185]: ZWO camera found. Choosing ZWO
May  2 04:59:20 allsky allsky.sh[2185]: Settings check done
May  2 04:59:20 allsky allsky.sh[2185]: CAMERA: ZWO
....
....
May  2 05:07:26 allsky allsky.sh[4430]: Settings check done
May  2 05:07:26 allsky allsky.sh[4430]: CAMERA: ZWO
May  2 05:07:26 allsky allsky.sh[4430]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
May  2 05:07:26 allsky allsky.sh[4430]: Starting allsky camera...
May  2 05:10:23 allsky allsky.sh[4430]: #033[32m ******************************************
May  2 05:10:23 allsky allsky.sh[4430]: #033[32m *** Allsky Camera Software v0.7 | 2020 ***
May  2 05:10:23 allsky allsky.sh[4430]: #033[32m ******************************************
May  2 05:10:23 allsky allsky.sh[4430]: #033[32mCapture images of the sky with a Raspberry Pi and an ASI Camera
May  2 05:10:23 allsky allsky.sh[4430]: #033[33mAdd -h or -help for available options
May  2 05:10:23 allsky allsky.sh[4430]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
May  2 05:10:23 allsky allsky.sh[4430]: #033[0mContributors:
May  2 05:10:23 allsky allsky.sh[4430]: -Knut Olav Klo
May  2 05:10:23 allsky allsky.sh[4430]: -Daniel Johnsen
May  2 05:10:23 allsky allsky.sh[4430]: -Yang and Sam from ZWO
May  2 05:10:23 allsky allsky.sh[4430]: -Robert Wagner

jcauthen78, 2021-05-04:
I've been having this same issue on a fresh install of Raspian Buster. 
Raspberry Pi 4b, 8gb
Updates & upgrades done. 
Fresh install of Allsky & GUI.
ASI120-MCS - firmware up to date.
Required triggers in config.sh are correct.
Second Pi4 8g setup about 1.5 months ago works as expected still (with same scripts), _although its a HQ Pi Cam._

Running all 3 scripts manually from command line **WORKS**. 
`/home/pi/allsky/scripts/timelapse.sh 20200503`
`./keogram ./images/20200503/ jpg ./images/20200503/keogram_20200503.jpg -fontsize 2`
_Removed the -rotate, didn't like how it made the image look._ 
`./startrails ./images/20200503/ jpg 0.33 ./images/20200503/startrails_20200503.jpg`

Checked for bad images, none found. 

I'll post the Allsky log file in the morning so its fresh.

paulfindus, 2021-05-04:
Glad to hear that I'm not alone with this...

jcauthen78, 2021-05-04:
Well - after looking at the allsky.log it may be a script conflict on my system (a keep-alive script) that may have been rebooting it. Disabled the script, we'll see what tomorrow brings.  I'm not too sure how to fake the allsky into running a run like its night time and seeing how it handles the endOfNight.sh.  When i _manually_ run endOfNight.sh, it runs just fine w/o errors.

paulfindus, 2021-05-05:
Thank you! Yes, I feel the same! But the error (german "Speicherzugriffsfehler" - Memory access error ("segfault") or so?!?!) always is present and this seems to break up everything...

jcauthen78, 2021-05-06:
The install was giving me some other headaches as well, so I decided I'd swap in a new memory card with a fresh install 
```
- Raspbian full (March 04 image). 
- sudo apt-get update
- sudo apt-get upgrade
- Allsky install
- Allsky Gui install
```
Last night's run completed successfully with all 3 files created. I'm not sure what was causing the headache on my end (bad memory card? corrupt install? gremlins?), but I'm gonna run with the fresh install and see how this second Allsky runs for me. 

Sorry I don't have any other suggestions, I know a complete fresh install isn't an ideal troubleshooting step

paulfindus, 2021-05-07:
hmmm..., I will try this gremlin tool ;- )))

paulfindus, 2021-05-07:
...tried to reproduce your gremlin procedure ;- )
03.04. image, update, upgrade, fully new install of allsky.
waiting for the darkness, so that the gremlins can do their work...
May the weather be with you

paulfindus, 2021-05-10:
so what my gremlins found out:
My cam ZWO 120Mc firmware is updated is USB 2.0.
After setting the USB bandwidth from default 40 to 100 all errors are gone and everything is running quiet, smooth and problemfree!!
All the pics are produced after the night ends...
Perhaps this helps someone else.
May the PI be with you

jcauthen78, 2021-05-10:
good catch! This also seems to have resolved similar problems ( including black daytime images, and freezing script) with my ASI120mc (current upgraded firmware) on my 2nd Pi4.

mlincett, 2021-06-06:
I am trying to troubleshoot a similar issue. Turning USB bandwidth to 100 initially solved the issue but after enabling the dark frame subtraction the generation stopped working again. I am not sure I understand what is that makes the keogram / startrail / timelapse generation fail in the first place and why some setting affecting the capture would make it fail.

I would like to provide more useful troubleshooting information but I do not know how to access the relevant log.