AllskyTeam/allsky #367: no keogram label on raspi4

loloderu, 2021-05-11:
Hi,
I cannot get labels on my keogram even running manually on a raspberry pi 4 with Raspbian GNU/Linux 10 (buster) OS.
Do you have any Idea why?
Regards

EricClaeys, 2021-08-20:
Would you please attach a sample keogram image so I can see what you are referring to?
Thanks - Eric

EricClaeys, 2021-10-16:
@loloderu,
Would you please attach a sample keogram image so I can see what you are referring to?
Thanks - Eric

loloderu, 2021-10-16:
Hi, sorry for not answering sooner.
It is all working now but I can't remember exactly what was causing the problem: I remember I ajusted the sie of the vertical line et label size, but not sure it if it was the reason why it was not working.

loloderu, 2022-01-25:
[image]

loloderu, 2022-01-25:
I have 2 allsky: one on Rpi3 is a PiHQ and gives me labels, and the other on RPi4 is an Zwoo ASI 224 MC and I can't get any labl with it (see the above picture), even with the most recent keogram.cpp
Do you have any Idea ?
Thanks for your help

EricClaeys, 2022-01-25:
@loloderu Are they running identical versions of Allsky?
In config/config.sh, what is the KEOGRAM_EXTRA_PARAMETERS set to on both machines?

loloderu, 2022-01-25:
Rpi3/PiHQ
--rotate 258 --font-color 255 255 255 --font-size 1.0 --font-line 2
Rpi4/ASI224MC
--rotate 270
However I tried for Rpi4 also the below option with same result
--rotate 270 --font-color 255 255 255 --font-size 1.0 --font-line 2

EricClaeys, 2022-01-26:
@loloderu Is the size of the keogram on the Pi 4 the same as your image size?
Also, since the resolutions of the cameras are different you may need different font settings.
Go overboard and set --font-size to 5 and --font-line to 4.

To test, create a directory called "test" in allsky/images and copy a dozen images from yesterday to the "test" directory.  Then run `allsky/scripts/generateForDay.sh-t test`.  Check the keogram in the "test" directory.

loloderu, 2022-01-26:
The size f the keograms are the same as the image that comes every minutes
Rpi3/PiHQ : 1500x1585
Rpi4/ASI224MC: 1500x1304
I tried --font-size to 5 and --font-line to 4 for the Pi4 but it made no differences

loloderu, 2022-01-28:
I finally found the reason: the images are not on the raspberry bu on distant windows machine. for some reason the file time is not interpreted correctly. If I use the option --parse-filename, the labes are reappearing!

EricClaeys, 2022-01-29:
@loloderu That's good to know.  I'll have to look at the code and see why it didn't output a warning message.

EricClaeys, 2022-02-20:
Closing.  I submitted a change to print a message when the date can't be determined.

closed 2022-02-20 by EricClaeys