AllskyTeam/allsky #368: Gain value limited to 100

willoubua, 2021-05-13:
Hello, 

I made my own Allsky based on ZWO ASI 120MC.
Everything works fine except one point, the Gain value is limited to 100. (from what i see in the metadata indicated on each picture).
I tried to set Gain and Max Gain value to 250 either from the GUI (Editor) or from the config file but nothing change, still 100 max.
I have disable of course the automatic gain option.

If someone allready met this situation and can help me...

Thank you
William

EricClaeys, 2021-07-08:
William,
in capture.cpp, where you find the line "ASIGetNumOfControls(CamNum, &iNumOfCtrl);", add the lines below, recompile, and restart allsky.  Look in the log for the gain MaxValue for your camera.
When done, either remove the code or change the first line to "if (0)"

if (1)	// this is really only needed for debugging
    {
    	// ECC: Display the control capabilities, not just the names, for debugging.
        printf("Control Caps:\n");
        for (i = 0; i < iNumOfCtrl; i++)
        {
            ASIGetControlCaps(CamNum, i, &ControlCaps);
            printf("- %s:\n", ControlCaps.Name);
            printf("   - MinValue = %ld\n", ControlCaps.MinValue);
            printf("   - MaxValue = %ld\n", ControlCaps.MaxValue);
            printf("   - DefaultValue = %ld\n", ControlCaps.DefaultValue);
            printf("   - IsAutoSupported = %d\n", ControlCaps.IsAutoSupported);
            printf("   - IsWritable = %d\n", ControlCaps.IsWritable);
            printf("   - ControlType = %d\n", ControlCaps.ControlType);
        }
    }

willoubua, 2021-07-17:
thank you, let me try and i'll come back to you.

William

EricClaeys, 2021-08-01:
William,
Were you able to try my suggestion?  If so, what did it show for gain?
Eric

willoubua, 2021-08-01:
Hello Eric, 

I didn't because to be honest i don't know how to recompile :-) i'm not a dev and i'm not used to work on linux.
I know how to edit the file and modify it but not the rest.
Maybe you could give me the command line to recompile ?

Thank you

William

EricClaeys, 2021-08-01:
William,
Open a command window (the black-ish icon on the left side of the tool bar). You should be in /home/pi.  Then:
     cd allsky
     make capture
Restart allsky like you normally do.  The log file can be very helpful.  It's normally in /var/log/allsky.log.

Let me know how it goes.

Eric

EricClaeys, 2021-08-20:
William,
The ASI120 camera has a max gain of 100.  There is nothing we can do about this.
Attached are its capabilities including the min, max, and default values.  If this answers your question, please close the issue.
Thanks - Eric

[ASI120MC_Caps.txt](https://github.com/thomasjacquin/allsky/files/7024454/ASI120MC_Caps.txt)

willoubua, 2021-08-21:
Ho damed...

I saw so much exemples with Gain arround 250 i though it was the same for all camera...

Thank you

closed 2021-08-21 by willoubua

EricClaeys, 2021-08-21:
I assume the max is 100 because the ASI120 is an older camera and the "mini" model is typically used as a guide camera.  My ASI178's max gain is much higher.
Eric