AllskyTeam/allsky #372: Startrail Error

nemopixels, 2021-05-20:
Hello,

The timelapse is not getting created automatically. So, I tried running it manually but it fails. I have tried changing the jpg values too. It doesn't help.

Rpi 4 + Rpi HQCam

Command:  ./startrails ./images/20210518/ jpg 0.3 ./images/20210518/startrails/test.jpg

Error:
[74/1031] image-20210518215221.jpg 0.386781
[75/1031] image-20210518215253.jpg 0.384044
[76/1031] image-20210518215341.jpg 0.382686
OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array') in binary_op, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp, line 225
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp:225: error: (-209) The operation is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array' in function binary_op

Aborted

Regards,
Nelson Moses

jcauthen78, 2021-05-20:
I've had a similar issue pop up when one or more of the images has been corrupted, or different physical dimensions.  my workaround for the corrupted images is to open up the folder in explorer, arrange by file size - most bad images will be below 100k (if any) - from there I delete the offending images, the process usually flows smoothly after that. 

There's an option in the endOfNight.sh to  have the script auto search/find for bad images during the main run. 

It's a thought, but may not be a solution to your current problem.

nemopixels, 2021-05-20:
Thank you very much! It works after deleting the corrupted images :)

closed 2021-05-20 by nemopixels