AllskyTeam/allsky #374: BeginQHYCCDLive failed

ptson92, 2021-05-28:
I try to install qhy camera at this link https://github.com/thomasjacquin/allsky/tree/qhy-camera
When I start allsky.sh, I get the error BeginQHYCCDLive failed.
Is there any way to fix this error?
Thanks very much!

ptson92, 2021-05-28:
******************************************
 *** Allsky Camera Software v0.6 | 2019 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options 

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>

Init SDK success!

Listing Attached Cameras:
connected to the first camera from the list,id is QHY5II-M-1047d910f33c7354
Open QHYCCD success!
Init QHYCCD success!
GetQHYCCDChipInfo success!
CCD/CMOS chip information:
Chip width 6.660000 mm,Chip height 5.320000 mm
Chip pixel width 5.200000 um,Chip pixel height 5.200000 um
Chip Max Resolution is 1280 x 1024,depth is 8
SetQHYCCDResolution success!
BeginQHYCCDLive failed

EricClaeys, 2021-08-20:
Thomas,
is anyone working on QHY support?  If so, who is it, so we can send ptson92 to that person?
Eric

ckuethe, 2021-09-17:
Funny you should mention that. I literally just started working on qhy support this evening. I'm trying to decide at what point I want to refactor all the `capture*cpp` files because right now the duplicated code is making my eyes bleed.

I think I can hold off on refactoring out some code into a utility library until qhy support is ready - then it'll be obvious what can move where...

EricClaeys, 2021-09-17:
Chris, I assume you know Thomas has a "qhy" branch?
The capture.cpp and capture_RPiHQ.cpp files more similar in version 0.8, and I have newer versions on my Pi that are even more similar, but still quite different.  I suspect it'll be a lot easier for ZWO and QHY versions to share a lot of code.

ckuethe, 2021-09-17:
I only discovered that after several hours of tinkering. I didn't like what I saw in the include directory but I'll see what's a good next step

closed 2021-10-16 by EricClaeys