AllskyTeam/allsky #378: sftp problems

Dave1680, 2021-06-12:
Hi Thomas,

I followed the video by: PatriotAstro 

And everything went well, the camera is connected and I have a live view and it creates image.jpg in the allsky directory on the pi and after setting the UPLOAD_IMG=true, it creates image-resize.jpg but the sftp [FAILS].

when I look in /var/log/allsky.log I am getting "cd: Fatal error: Host key verification failed"

Any suggestions as to what the problem might be?

Thanks in advance,

Dave.

--------------------------------------------------------------------------------------------------------------------------------------

ftp-settings.sh

#!/bin/bash

# FTP/SFTP/S3 settings
PROTOCOL='sftp'
USER='xxxxxxxxxxxxxxxx'
PASSWORD='xxxxxxxxxxxxxxxx'
HOST='accessxxxxxxxx.webspace-data.io'
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'

/var/log/allsky.log

Jun 11 14:35:55 pisky allsky.sh[10791]: Resizing
Jun 11 14:35:55 pisky allsky.sh[10791]: Uploading
Jun 11 14:35:56 pisky allsky.sh[10791]: cd: Fatal error: Host key verification failed.
Jun 11 14:35:59 pisky allsky.sh[10791]: put: image-resize.jpg: Fatal error: max-retries exceeded

The allsky directory on the server is in /allsky/  -  sat under the root directory so on the host at ./allsky

Dave1680, 2021-06-12:
Fixed the problem.

I had to ssh to the raspberry PI then try to ssh to my host as below:

ssh -p 22 username@host

the host then replied with a yes/no which you answer yes.

The uploads then worked.

Dave.

minnican, 2021-07-22:
Thanks Dave... You just saved me a whole lot of pain. While transferring my website from commercial host to Raspberry Pi 4 local web server I just couldn't figure out how to get the Allsky Raspberry Pi to sftp to my new server despite having just used sftp to upload my website files to the server using Filezilla. Your solution worked a treat!

EricClaeys, 2021-08-20:
Dave,
Glad it worked.  I'm keeping track of problems and solutions so I can enter into a "troubleshooting" guide.
Would you please close this issue?
Thanks - Eric

closed 2021-08-21 by Dave1680