AllskyTeam/allsky #380: Getting location and time from NMEA/GPSD

tlkw79, 2021-06-15:
Hi,

Is it possible for AllSky to update location and time with GPS/GPSD instead of fill up by ourselves?

Regards,
Tom

EricClaeys, 2021-08-20:
Tom,
Do you have a GPS unit that saves data to a file?  If the file is in a standard format (rather than a format specific to one brand of GPS), then a script could be written to update the AllSky configuration files with the latitude and longitude, and I suppose with the time as well.
I'm not sure it's worth the effort though, since not many people have a GPS connected to their Pi.

Eric

closed 2021-09-05 by EricClaeys

wglogowski, 2021-09-05:
Yes, I subscribed 

"Equipped with his five senses, man explores the universe around him and calls the adventure Science."
Edwin Hubble