AllskyTeam/allsky #381: keogram orientation on raspberrypi 3B+ with RPIHQ camera

rodrigoluk, 2021-06-18:
I've got most of the software running on a new build, including the GUI. Great stuff! Thank you!
 
Capturing images, generating the timelapse (resizing the images from the RPIHQ camera) all work fine. I'm not so sure about startrails yet. The keogram gets generated but it is a vertical image (242x3040) - Any ideas on how to resolve this?

Thanks!

AndreasLMeg, 2021-07-17:
#387
some facts about keogram

rodrigoluk, 2021-07-18:
Many thanks for replying. I didn't quite get that the keogram would look like this for just a few hours of exposures. It should have been clear to me from the example in the readme that it represent a full day. I've got everything working and I'm fine tuning stuff. 
Thank you!

EricClaeys, 2021-08-20:
rodrigoluk,
If Andreas' post answered your question, would you please close the issue?
Thanks - Eric

closed 2021-08-21 by rodrigoluk